hardbyte/python-can #636: code not working after updating from 3.1.1. to higher versions

philippherget, 2019-07-09:
Related to #593

I am using a Raspberry Pi Zero W with a Waveshare rs485 can board and raspbian installed. The python-can version was 3.1.1. 

Since updating to 3.2.0 I run into several issues. First of all 'is_extended_id' causes an error and seems to be unknown. I figured out it's working when replaced with 'extended_id'.
`TypeError: __init__() got an unexpected keyword argument 'is_extended_id'`

'Send_periodic' isn't working anymore.
`Attribute Error: 'SocketcanNative_Bus' object has no attribute 'send_periodic'`

Additionally there is an issue with 'modify_data'.
`Attribute Error: 'NoneType' object has no attribute 'modify_data'`

I tried all the releases mentioned in #593 and also 3.3.0. If I go back to 3.1.1 my code doesn't work anymore and I can't figure out what might be the issue. When installing it finishes the installation but during the installation mentions the windows-curses issue.

Code I am running:

`import os
import time
import can

os.system("sudo /sbin/ip link set can0 up type can bitrate 250000")

bus = can.interface.Bus(bustype='socketcan', channel='can0', bitrate=250000)

logging.basicConfig(level=logging.INFO)

def Simulation(bus):
    msg = can.Message(arbitration_id=0x18F00400, data=[0, 0, 0, 0xF4, 0x01, 0, 0, 0], is_extended_id=True)
    task = bus.send_periodic(msg, 1)
    time.sleep(1)
    msg.data[3] = (0x1F4 & 0xFF)
    msg.data[4] = ((0x1F4 & 0xFF) >> 8)
    task.modify_data(msg)
    time.sleep(1)
`

Everything was working as intended before updating.

mikado98765, 2019-07-09:
Not sure where the send_periodic error is coming from, but since that gives an error then you won't get a task object back in `task = bus.send_periodic(msg, 1)`. Therefore, when you try to call modify_data on a null object, you'll get that error

mikado98765, 2019-07-09:
The send_periodic error makes me think that there was an error with the install. Does something like bus.send(msg) give expected results or any errors?

felixdivo, 2019-07-10:
Can you print `can.__version__`? The problem with `is_extended_id` is weird, since that is the non-deprecated way to do it. Could you also print `import sys; print(sys.version)`?

philippherget, 2019-07-15:
sys.version returns:

```
3.5.3 (default, Sep 27 2018, 17:25:39)
[GCC 6.3.0 20170516]
```

can.__version__ returns:

`AttributeError: module 'can' has no attribute '__version__'`

If I replace bus.send_periodic(msg, 1) with bus.send(msg) it seems to be working, but returning the same modify_data error later on.

felixdivo, 2019-07-15:
It is really strange that `can.__version__` does not work ...
Are you really sure that it is pointing to the correct installation? In may case, `can.__file__` shows that the library is loaded from `/home/username/.local/lib/python3.6/site-packages/can/__init__.py`.

philippherget, 2019-07-15:
`can.__file__` returns in my case:

`/usr/lib/python3/dist-packages/can/__init__.py`

The content of `__init__.py` is:

```
"""
can is an object-orient Controller Area Network interface module.
"""
import logging
log = logging.getLogger('can')

rc = dict(channel=0)

class CanError(IOError):
   pass

from can.CAN import BufferedReader, Listener, Logger, Printer, CSVWriter, SqliteWriter, ASCWriter, set_logging_level
from can.message import Message
from can.bus import BusABC
from can.notifier import Notifier
from can.broadcastmanager import send_periodic, CyclicSendTaskABC, MultiRateCyclicSendTaskABC
from can.interfaces import interface

```

felixdivo, 2019-07-15:
I think that version is ancient ... It now looks somewhat like [this](https://github.com/hardbyte/python-can/blob/develop/can/__init__.py). The version information [was added more than two years ago](https://github.com/hardbyte/python-can/commit/af05d661e586b19a4e4fcfdc4295bb9d0f6b8441#diff-9479df40d1b902b423163d3ea3c287de), and already back then, the `__init__.py` was a lot larger/different.

felixdivo, 2019-07-15:
What command did you use to install this package?

philippherget, 2019-07-15:
I used `pip install python-can`.

felixdivo, 2019-07-15:
That is probably installing it for Python 2, right? Please use `pip3` or check the above code parts in Python 2 (`python2`).

philippherget, 2019-07-15:
You are right. I forgot to add the alias to set pip = pip3. I will try if this fixes the issue.

hardbyte, 2019-07-27:
Please reopen if you're still having issues with the new v3.3.1 release.

closed 2019-07-27 by hardbyte

philippherget, 2019-07-29:
Working fine now/again. Thank you for all your help.