hardbyte/python-can #648: Log rotation

HellDeveloper91, 2019-07-19:
Hello,

We are testing can devices and we use the ASCWritter and BLFWritter to log all the can traces as evidences. The problem is that the log file created for long duration tests are so big that are unusable.
Looking in the documentation we don't find any configuration to make the logs rotate when the file reaches X size. There is any way to do it or is it any plans to add it?

Thank you.

christiansandberg, 2019-07-19:
You could implement that logic yourself by calling the stop() method and creating a new log file. If you’re using the Notifier class you may need to implement the background thread yourself or removing and adding listeners in the main thread.

felixdivo, 2019-07-24:
I think this would be a bit to much for this library, but would be a perfect fir for a library building on this one.

HellDeveloper91, 2019-07-24:
I will do a workaround doing something like start and stoping as you say.
Also, I've seen that on the file listeners you are using a file directly. Maybe using the logging module with a RotatingFileHandler could be better?

felixdivo, 2019-07-25:
*That* seems cool.

closed 2019-07-27 by hardbyte