hardbyte/python-can #649: Can't install python-can without internet access

chrisoro, 2019-07-19:
In my organization we have computers that are not connected to the internet. We have an artifactory setup as remote repository for the official pypi. In short: we can do pip install python-can==3.1 --index-url="path-to-our-mirror", that uses the mirrored artifactory which then asks the real pypi to download python-can.

A few months ago, setup_requires=["pytest-runner"], has been added to the setup.py which skips pip and downloads pytest-runner directly. This won't work if you can't access the hoster.

This has been (partially) fixed here: https://github.com/hardbyte/python-can/pull/633 by installing pytest-runner only when you want to. Maybe a better fix would be to shift it to install_requires? Anyways it is scheduled for your v4 release but that is please a few weeks out.

Can you guys release a bugfix 3.3.1 or 3.3.0.1 (or whatever) that includes this fix? Without it, it's not possible to install python-can>3.1 without internet access

hardbyte, 2019-07-24:
Hi @chrisoro, I've released v3.3.1 now with that fix backported. Please reopen if that doesn't fix it for you.

closed 2019-07-24 by hardbyte

chrisoro, 2019-07-26:
Thanks @hardbyte, works fine. Thank you!