hardbyte/python-can #660: .modify_data() only changing one byte

philippherget, 2019-07-30:
Related to #636 

So basically after updating to 3.3.1 everything is working fine. But it seems like the modify_data() command no longer works. With the code below I could always change both bytes of the message, but since the update to 3.3.1 only the third byte is changed/first command is executed. The fourth byte is ignored, although the change of two bytes worked fine before (version 3.0.0 used).

```
import os
import time
import can

os.system("sudo /sbin/ip link set can0 up type can bitrate 250000")

bus = can.interface.Bus(bustype='socketcan', channel='can0', bitrate=250000)

logging.basicConfig(level=logging.INFO)

def Simulation(bus):
    msg = can.Message(arbitration_id=0x18F00400, data=[0, 0, 0, 0xF4, 0x01, 0, 0, 0], is_extended_id=True)
    task = bus.send_periodic(msg, 1)
    time.sleep(1)
    msg.data[3] = (0x1F4 & 0xFF)
    msg.data[4] = ((0x1F4 & 0xFF) >> 8)
    task.modify_data(msg)
    time.sleep(1)

```

karlding, 2019-07-30:
Perhaps I'm missing something, but looking at your code, it appears that your masked result for `msg.data[3]` is the same as what you've initially set in your data.

```python
data=[0, 0, 0, 0xF4, 0x01, 0, 0, 0]

# Assert that the third byte (0-indexed) is equal to 0xF4
assert data[3] == (0x1F4 & 0xFF)

# Assert that the fourth byte (0-indexed) should be 0x00
assert 0x00 == ((0x1F4 & 0xFF) >> 8)
assert data[4] != ((0x1F4 & 0xFF) >> 8)
```

So only one byte (byte 4) should be changing.

```
  vcan0  18F00400   [8]  00 00 00 F4 01 00 00 00
  vcan0  18F00400   [8]  00 00 00 F4 00 00 00 00
                                     ^^
```

Can you confirm that this code is correct? You said in the bug report that:

> only the third byte is changed/first command is executed. The fourth byte is ignored,

which doesn't seem to line up with what the code is doing.

philippherget, 2019-07-31:
> Perhaps I'm missing something, but looking at your code, it appears that your masked result for msg.data[3] is the same as what you've initially set in your data.

You are right but that is just the first modify_data task. There are more calls following that should insert different values but I wanted to keep my issue simple. The problem that only the third byte is changed remains independent of the value used.

> Can you confirm that this code is correct?

Yes this code is correct but mine is still executed as before and is only changing the third byte.

The point is I try to send J1939 conformal messages on the bus so the given message is transmitted on two bytes. With my code I try to distribute and transfer a selected value (e.g. 500 = 0x1F4) over two bytes. I did it the same way on 3.0.0 and it was working fine back then. The ID 18F00400 represents the rpm of a machine and the corresponding value is transferred to two bytes.

karlding, 2019-08-01:
Are you also able to reproduce this using the virtual (`vcan`) SocketCAN driver?

```bash
# Load kernel module
sudo modprobe vcan

# Bring up a vcan interface
sudo ip link add dev vcan0 type vcan
sudo ip link set up vcan0

# View data being sent on vcan0 interface
candump vcan0
```

philippherget, 2019-08-05:
I reproduced the code using the vcan SocketCAN driver. I added some `modify_data` blocks to have different values changing so the codeblock looks like this now:

```
def Simulation(bus):
    msg = can.Message(arbitration_id=0x18F00400, data=[0, 0, 0, 0, 0, 0, 0, 0], is_extended_id=True)
    task = bus.send_periodic(msg, 1)
    time.sleep(1)
    msg.data[3] = (0x1F4 & 0xFF)
    msg.data[4] = ((0x1F4 & 0xFF) >> 8)
    task.modify_data(msg)
    time.sleep(1)
    msg.data[3] = (0x3E8 & 0xFF)
    msg.data[4] = ((0x3E8 & 0xFF) >> 8)
    task.modify_data(msg)
    time.sleep(1)
    msg.data[3] = (0x5DC & 0xFF)
    msg.data[4] = ((0x5DC & 0xFF) >> 8)
    task.modify_data(msg)
    time.sleep(1)
```

candump shows the following:

```
vcan0  18F00400   [8]  00 00 00 00 00 00 00 00
vcan0  18F00400   [8]  00 00 00 F4 00 00 00 00
vcan0  18F00400   [8]  00 00 00 E8 00 00 00 00
vcan0  18F00400   [8]  00 00 00 DC 00 00 00 00
```

I still can't figure out why it only changes one byte and sets the other one permanently to zero.

karlding, 2019-08-05:
Given that code, your output is expected, since you're masking the lower 8 bits and then right shifting. Like I previously mentioned, you're setting byte 4 to `0`, so it's "permanently set to 0".

Did you intend to perform the right shift first, and then mask?

```diff
index 7dfe920..038ec8f 100644
--- 1/issue-660.py
+++ 2/issue-660.py
@@ -11,17 +11,17 @@ def Simulation(bus):
     time.sleep(1)

     msg.data[3] = (0x1F4 & 0xFF)
-    msg.data[4] = ((0x1F4 & 0xFF) >> 8)
+    msg.data[4] = ((0x1F4 >> 8) & 0xFF)
     task.modify_data(msg)
     time.sleep(1)

     msg.data[3] = (0x3E8 & 0xFF)
-    msg.data[4] = ((0x3E8 & 0xFF) >> 8)
+    msg.data[4] = ((0x3E8 >> 8) & 0xFF)
     task.modify_data(msg)
     time.sleep(1)

     msg.data[3] = (0x5DC & 0xFF)
-    msg.data[4] = ((0x5DC & 0xFF) >> 8)
+    msg.data[4] = ((0x5DC >> 8) & 0xFF)
     task.modify_data(msg)
     time.sleep(1)
```

Running this gives me:

```
$ candump vcan0
  vcan0  18F00400   [8]  00 00 00 00 00 00 00 00
  vcan0  18F00400   [8]  00 00 00 00 00 00 00 00
  vcan0  18F00400   [8]  00 00 00 F4 01 00 00 00
  vcan0  18F00400   [8]  00 00 00 E8 03 00 00 00
  vcan0  18F00400   [8]  00 00 00 DC 05 00 00 00
```

philippherget, 2019-08-05:
That's exactly what I want as an output. Your explanation also makes sense to me now. It's a bit embarrassing but now it works as intended. However, I find it very astonishing that my code worked as desired with the 3.0.0. release. Nevertheless many thanks for your help!

karlding, 2019-08-09:
I'm going to close this issue. Please reopen if necessary!

closed 2019-08-09 by karlding