hardbyte/python-can #661: Evaluate switching over to Azure Pipelines (or running in parallel)?

karlding, 2019-07-31:
Is switching Travis CI over to Azure Pipelines (or running it in parallel) for the CI checks something that we would be interested in?

According to Microsoft's [original announcement](https://azure.microsoft.com/en-us/blog/announcing-azure-pipelines-with-unlimited-ci-cd-minutes-for-open-source/):

> Azure Pipelines provides unlimited CI/CD minutes and 10 parallel jobs to every open source project for free. All open source projects run on the same infrastructure that our paying customers use. That means you’ll have the same fast performance and high quality of service.

At the very least, it might be worthwhile to evaluate their macOS offerings, since it appears they provide Mojave/High Sierra agents.

hardbyte, 2019-07-31:
I've been trying out Azure Pipelines in another pure Python project which has been a mostly positive experience, I didn't try the macOS agents though. There are a few Microsoft/Powershell idioms that are a bit strange. Take a look at these links:

- [results of a build](https://dev.azure.com/data61/Anonlink/_build/results?buildId=165&view=logs)
- [release to PyPi pipeline](https://dev.azure.com/data61/Anonlink/_releaseProgress?_a=release-pipeline-progress&releaseId=15)
- [`azurePipeline.yml`](https://github.com/data61/clkhash/blob/master/azurePipeline.yml)
- [`.azurePipeline/wholeBuild`](https://github.com/data61/clkhash/blob/master/.azurePipeline/wholeBuild.yml)

hardbyte, 2020-11-25:
Given Github workflows offer slightly less friction I think we should aim to use them instead.

closed 2020-11-25 by hardbyte