hardbyte/python-can #668: Add an overview over what interface is supported on which platform

felixdivo, 2019-08-03:
Additional, the list of interface can be found at multiple locations, it would be a good point to maybe clean it up.

I can do this in a few days/weeks, if nobody is eager to claim this task. ;)

christiansandberg, 2019-08-20:
It would be nice to also list which interfaces support which features. Maybe in table form but it might get pretty big...

felixdivo, 2019-08-30:
This is exactly what I thought about. ;) We can probably make it compact enough somehow.

What features would be important?

- Basic frames
- Error frames
- Remote frames
- cyclic sending (always supported, but maybe differentiate between Userspace, Kernel and hardware?)
- supported OS (Win, Linux, Mac?)
- what extras need to be installed with pip

Such a table would also be helpful for the file I/O classes.

christiansandberg, 2019-08-30:
* Detect available configs
* Low level filtering
* Receive own messages
* Absolute timestamps
* Blocking send
* Listen only/state
* Flush tx buffer
* Reset