hardbyte/python-can #676: Stop logging messages to command prompt

sreedevi101, 2019-08-07:
Hi,

I am starting and stopping the py-can notifier from a main program. Once the main program completes running, messages from py-can are printed in the command prompt. How can I stop this as I don't want the messges to be printed in the command prompt?

Py-can program: 

`self.bus` = `can.Bus(interface='neovi',channel='1',receive_own_messages=True)`

`self.notifier = can.Notifier(self.bus, [self.parse_message])`
...
...

`self.notifier.stop(timeout=0)`

mikado98765, 2019-08-15:
What exactly does `self.parse_message` in its stop method? The listener itself doesn't print things to the command prompt but its possible that `stop` method of `parse_message` has print statements in it, or it flushes some buffer, etc.

sreedevi101, 2019-08-15:
Thank you for your reply. Let me give more details about the program.

`class ReceiveCAN()` is called from the main application (which is executed from command prompt) and contains methods - `start_notifier(self)`, `parse_message(self, SingleCanFrame)`, `stop_notifier(self)`

`parse_message` method is called by notifier whenever a new message is received.
`parse_message` prints the message id and data to a file.

There are no print statements and messages are properly written to the file.

Messages are printed in the command prompt and the following warning is also printed: 

> "Rx message buffer overflow "

mikado98765, 2019-08-15:
If all the messages are getting printed at once along with the warning, it could be because they are part of the error (possibly, not too sure on this one)? Which version of python-can are you using? The reason I ask is because I believe `can.Notifier` takes a bus and a list of `can.Listener` objects as parameters, but it looks like you are just passing a method instead of a listener object. Surprisingly, it looks like this would work until you try to call stop on the Notifier (https://python-can.readthedocs.io/en/master/_modules/can/notifier.html#Notifier.stop). Because `parse_message` doesn't have a stop attribute, it won't get called which I imagine gives some kind of undefined behavior

sreedevi101, 2019-08-21:
Hi,
I am using python-can version 3.1.1.
Please suggest me how can I change my `can.Notifier` to stop printing all messages in command prompt. I am calling `notifier.stop(timeout=0)` at the end.

felixdivo, 2019-09-04:
You could simply iterate over all messages directly (i.e. to polling):

```python
for message in bus:
    treat_message(message)
```

For us to help, a more complete program along with the log would be helpful.

closed 2019-09-20 by felixdivo