hardbyte/python-can #680: PCAN interace is broken

bmeisels, 2019-08-08:
I am testing some code with pcan and updated to the head of the development branch (previously was on 3.2.0).

When trying to create a PcanBus i get the following error:
```
c:\git_repos\python-can\can\interfaces\pcan\pcan.py in __init__(self, channel, state, bitrate, *args, **kwargs)
    185
    186         if state is BusState.ACTIVE or state is BusState.PASSIVE:
--> 187             self.state = state
    188         else:
    189             raise ArgumentError("BusState must be Active or Passive")

c:\git_repos\python-can\can\interfaces\pcan\pcan.py in state(self, new_state)
    464         if new_state is BusState.ACTIVE:
    465             self.m_objPCANBasic.SetValue(
--> 466                 self.m_PcanHandle, PCAN_LISTEN_ONLY, PCAN_PARAMETER_OFF
    467             )
    468

c:\git_repos\python-can\can\interfaces\pcan\basic.py in SetValue(self, Channel, Parameter, Buffer)
    823         """
    824         try:
--> 825             if Parameter in {PCAN_LOG_LOCATION, PCAN_LOG_TEXT, PCAN_TRACE_LOCATION}:
    826                 mybuffer = create_string_buffer(256)
    827             else:

TypeError: unhashable type
```

It seems 19b4ddf940b83b393268c7f424600acd6af59496 has caused a bug as the type of Parameter is ```c_ubyte``` which is not hashable.

My python version string is "3.7.2 (tags/v3.7.2:9a3ffc0492, Dec 23 2018, 22:20:52) [MSC v.1916 32 bit (Intel)]"

If i am correct and this is a bug i suggest reverting these changes.

karlding, 2019-08-08:
@bmeisels that sounds correct, since `c_ubyte` is a `CData` object, and `CData` objects are mutable, so they can't be hashed.

Can you test that reverting this fixes the problem, and then submit a patch?

bmeisels, 2019-08-08:
@bmeisels I tested it earlier today and saw that reverting these changes solves the problems.
I will submit a PR on Monday.

commit 807fc54d (2019-08-13): resolves #680
closed 2019-08-13 by hardbyte