hardbyte/python-can #681: Linux Python 3+ does not allow for RTR

RISKYY, 2019-08-08:
Environment: Python 3.6
OS: Linux Ubuntu
Chipset: x86 Lenovo Laptop
Error: AttributeError: 'int' object has no attribute 'value'

https://github.com/hardbyte/python-can/blob/1f0fb8b680a24e727a6995949c4f41d6251abd56/can/interfaces/pcan/pcan.py#L389

It appears that on Linux msgType doesn't have a value, whereas on OSX it does. If you remove ".value" on linux it runs and works fine. Though, if you run the same removed .value on OSX it errors.

craffert0, 2019-08-14:
Fixed in https://github.com/hardbyte/python-can/pull/659

hardbyte, 2019-08-15:
Closed because this has been addressed in `develop`, but I have opened #686 to backport to a 3.x release.

closed 2019-08-15 by hardbyte