hardbyte/python-can #683: Problems with player.py

Ralph-Ne, 2019-08-12:


mikado98765, 2019-08-14:
If they aren't proprietary, could you post the contents of FILE_TO_PLAY, or what format its in?

Ralph-Ne, 2019-08-15:
[CAN_JD670i_j1939_2017-08-31_13-06-00.blf.zip](https://github.com/hardbyte/python-can/files/3505400/CAN_JD670i_j1939_2017-08-31_13-06-00.blf.zip)

mikado98765, 2019-08-15:
I've tried the file on a different interface and it appears to work. The smaller values of g probably mean that messages are getting sent too fast, filling up the tx buffer, as the error suggests. Are you using a virtual or physical device? Possibly helpful link: http://socket-can.996257.n3.nabble.com/increase-buffer-size-td2024.html

Ralph-Ne, 2019-08-15:
"ip link set can0 up txqueuelen 1000" is set in both cases, "slcand -o -c -f -s5 /dev/ttyUSB0 can0" to bind the device, so that "can0" appears when I do "ip addr"
The device, which analyses the data from the canplayer via the USB-adapter is a physical one.
If I use the player.py-2.1.1 it works fine, but player.py-3.3.1  is much to slow to play the data in the *.blf file in time.

christiansandberg, 2019-08-15:
Does the log contain error frames? One difference with the BLF parser is that it recognizes error frames in the log and will therefore add them to the socketcan transmit queue, possibly faster than can be processed. Obviously you can’t send error frames on a physical bus, but it’s possible on a virtual.

Ralph-Ne, 2019-08-16:
Yes, the log contains error frames. It's a normal log from a farm machine with error and "good data" frames. But why they are treated particulary?

christiansandberg, 2019-08-16:
Error frames are treated like regular frames basically, but what happens when trying to send an error frame is interface dependent. Mostly they will probably be discarded by the driver.

I’m thinking however that the player should by default *not* send error frames, but it could be enabled by a switch if desired.

Ralph-Ne, 2019-08-19:
Which switch I have to use to enable error frames?

felixdivo, 2019-09-04:
Seems like it would be `--error-frames`.

PR #690 (2019-09-04): Ignore error frames in can.player by default
They are usually not of interest and may fill up the transmit buffer.

Fixes #683
commit 1ac5510f (2019-09-04): Ignore error frames in can.player by default

Fixes #683
closed 2019-09-04 by felixdivo