hardbyte/python-can #692: KWP Fast Init

skast96, 2019-08-26:
Hello everyone, 

it`s not exactly an issue, but I was trying to establish a connection to a Car with KWP Fast and it failed. Has anybody tried that yet with this library?

If it helps this is my init message.
channel is can0
bitrate is set to 10400
`  msg = can.Message(arbitration_id= 0x7DF,
                          data=[0xC1, 0x33, 0xF1, 0x81],
                          is_extended_id=False)
`

felixdivo, 2019-09-04:
What interface where you using (i.e. how is your `bus` instance configured?) How exactly did it fail?

skast96, 2019-09-10:
I am running my config on a raspberry pi 3 Model B and it with a Sparkfun CanShield via SPI. 
https://www.sparkfun.com/products/13262

The bus is configured with can0 at 8 MHz with a baud rate of 10400.

felixdivo, 2019-09-10:
Okay, I could have been more explicit: What code are you using to setup your `bus` instance and what exception/stacktrace did your receive?

skast96, 2019-09-18:
Did you mean the code below ?
```python
  can.rc['interface'] = 'socketcan'
            can.rc['channel'] = 'can0'
            can.rc['bitrate'] = 10400
            self.bus = can.Bus()
```

karlding, 2019-09-30:
KWP refers to the protocol layer, so if the underlying physical layer that KWP is implemented on is CAN, then you will be able to receive CAN messages. Note that a K-line physical layer is different from using the CAN bus as the physical layer.

According to the [Wikipedia article](https://en.wikipedia.org/wiki/Keyword_Protocol_2000), when implemented on a CAN physical layer, KWP typically uses ISO-TP (which is provided via packages like [can-isotp](https://pypi.org/project/can-isotp/)). If your car actually uses a CAN physical layer to implement KWP, then there's no reason why `python-can` shouldn't work.

karlding, 2019-10-07:
Closing this out as there's nothing actionable here.

Please reopen if needed!

closed 2019-10-07 by karlding