hardbyte/python-can #695: Possible bug in viwer.py

felixdivo, 2019-09-04:
Could it be, that in `can/viewer.py` around line 169 there is an error? The excerpt of the relevant method follows. The very last else statement should be indented one level more, shouldn't it?

```python
    @staticmethod
    def unpack_data(
        cmd, cmd_to_struct, data
    ):  # type: (int, Dict, bytes) -> List[Union[float, int]]
        if not cmd_to_struct or not data:
            # These messages do not contain a data package
            return []

        for key in cmd_to_struct:
            if cmd == key if isinstance(key, int) else cmd in key:
                value = cmd_to_struct[key]
                if isinstance(value, tuple):
                    # The struct is given as the fist argument
                    struct_t = value[0]  # type: struct.Struct

                    # The conversion from raw values to SI-units are given in the rest of the tuple
                    values = [
                        d // val if isinstance(val, int) else float(d) / val
                        for d, val in zip(struct_t.unpack(data), value[1:])
                    ]
                else:
                    # No conversion from SI-units is needed
                    struct_t = value  # type: struct.Struct
                    values = list(struct_t.unpack(data))

                return values
        else:
            raise ValueError("Unknown command: 0x{:02X}".format(cmd))
```

Lauszus, 2019-09-04:
@felixdivo no that is not a bug. It's raising the exception if the for-loop completes without ever finding a match. See: http://book.pythontips.com/en/latest/for_-_else.html#else-clause.

felixdivo, 2019-09-04:
Okay yeah, I thought it should raise when something could not be found in `key`, but it's okay if that is intended. 👍

closed 2019-09-04 by felixdivo