hardbyte/python-can #698: Release 3.3.2 not in PyPI

jluttine, 2019-09-07:
I noticed there's a release 3.3.2: https://github.com/hardbyte/python-can/releases/tag/3.3.2

But currently the most recent version in PyPI is 3.3.1: https://pypi.org/project/python-can/

Could you release also in PyPI?

aquatsr, 2019-09-08:
Maybe pypi rejected it because the version still says `3.2.0` ?
```
~/Downloads/python-can-3.3.2$ grep -ir "__VERSION__" .
./can/__init__.py:__version__ = "3.2.0"
```

hardbyte, 2019-09-10:
@jluttine thanks for pointing this out - looks like I tagged the develop branch instead of master. I've just re-tagged and creating a new release in GitHub, 🤞 travis-ci should be building and uploading to PyPi soon.

@aquatsr you can see the latest `__init__.py` in master has the correct version: https://github.com/hardbyte/python-can/blob/master/can/__init__.py#L11

hardbyte, 2019-09-10:
v3.3.2 is now available in PyPi - https://pypi.org/project/python-can/3.3.2/

closed 2019-09-10 by hardbyte