hardbyte/python-can #702: Ixxat driver ignores UniqueHardwareID and [r|t]xFifoSize from configuration

cranium-xx, 2019-09-16:
The Ixxat canlib ignores UniqueHardwareID, rxFifoSize and txFifoxSize from the configuration file. 
The reason is that the driver requests the configuration with CamelCase but it has to use lower case.

felixdivo, 2019-09-20:
Can you provide a pull request solving the problem?

cranium-xx, 2019-09-20:
Of course, I can. I just need the rights to create an branch for it. How do I get them?

felixdivo, 2019-09-20:
Cool! The usual workflow is to fork (upper right corner of the page) and then apply the changes to the forked develop branch of `cranium-xx/python-can`. Then in your fork, you should see a button for merging it back into `hardbyte/python-can`.

felixdivo, 2021-04-14:
@cranium-xx This is very old. Do you plan on providing a patch?

cranium-xx, 2021-04-20:
Hi,
I just provided a patch.

felixdivo, 2021-04-21:
Great I'll have a look over there.

PR #1062 (2021-06-06): Preserve capitalization when reading config files
This change was discussed in #1020, where @cranium-xx worked on a fix for #702. Previously, all parameters were converted to lower case, and some parameters with upper case naming could therefore not be set. This is fixed by this PR.

Fix #702.
closed 2021-06-06 by felixdivo