hardbyte/python-can #707: Let the notifier notify about error frames

tamenol, 2019-10-11:
Hello,

I have to monitor a PCAN bus. I have written an asyncio script to read all the ID's on the bus and calculate the throughput on the bus. However I also have to count the error messages, but they aren't passed to the notifier. 
How can I do this? In PCAN-View I can see error messages ariving, ex. type=Error and RxERR=132. I'd like to pass these to my script so I can count.  At the moment the error messages get surpressed or the error 'Bus Error: ... heavy/light bus' (https://github.com/hardbyte/python-can/blob/develop/can/interfaces/pcan/pcan.py#L320) gets printed as a warning.

Thanks

tamenol, 2019-10-15:
`import asyncio
import can
import logging
import winreg

class PCAN(object):

    RX_SDO = 0x600
    TX_SDO = 0x580
    RX_PDO = 0x200
    TX_PDO = 0x180

    _count = 0
    _bit_count = 0
    _error_count = 0
    count = 0
    bitcount = 0
    error_count = 0

    def _count_msgs(self, msg):
        if msg.is_error_frame:
            self._error_count += 1
            print("error: {}".format(msg))
        else:
            self._count += 1
            self._bit_count += 46 + msg.dlc*8

    def __init__(self):

        self.logger = logging.getLogger("CAN")
        self.logger.info("Initializing CANbus")

        can.set_logging_level("debug")

        if self.check_key():
            self.bus = can.Bus(channel="PCAN_USBBUS1", bustype="pcan")
            self.listener = []
            self.loop = asyncio.get_event_loop()
            self.notifier = can.Notifier(self.bus, self.listener, loop=self.loop)
        else:
            raise Exception("NoDriverKeyERROR")

    def check_key(self):

        key_exists = True
        aReg = winreg.ConnectRegistry(None, winreg.HKEY_LOCAL_MACHINE)

        try:
            aKey = winreg.OpenKey(aReg, r"SOFTWARE\PEAK-System\PEAK-Drivers")
            winreg.CloseKey(aKey)
        except WindowsError:
            key_exists = False

        winreg.CloseKey(aReg)

        if key_exists:
            self.logger.info("Found PEAK-Drivers registry key")
            return True

        else:
            self.logger.warning("Unable to find PEAK-Drivers registery key. Install the driver.")
            return False

    async def throughput(self):

        self.logger.info("Counting")
        self.bus.set_filters()
        self._count = 0
        self._bit_count = 0
        self._error_count = 0
        self.notifier.add_listener(self._count_msgs)
        await asyncio.sleep(1)
        self.notifier.remove_listener(self._count_msgs)
        count = self._count
        bitcount = self._bit_count / 1000.
        error_count = self._error_count
        print("bitcount: {}\t\terrors: {}".format(bitcount, error_count))
        return (bitcount, error_count)

    def cleanup(self):
        """
        A function to clean everything up.
        """
        self.notifier.stop()
        self.bus.shutdown()
        self.logger.info("Stopped notifier; Bus shutdown")

    def __del__(self):

        self.logger.info("Destructor called")
        try:
            self.cleanup()
        except:
            pass

try:
    loop = asyncio.get_event_loop()
    cn = PCAN()
    while True:
        loop.run_until_complete(cn.throughput())
finally:
    loop.close()`

christiansandberg, 2019-10-15:
Error frames should be supported by pcan. The bus load warnings that PCAN produces are neither error frames nor exceptions so it there is no good way to report these. In your application however you probably don’t want to count these as they are completely internal events.

tamenol, 2019-10-16:
Hey, thanks for your time and reply. Yeah, I figured as much, I tried to look at the code to see where in the module the error gets thrown to see if some messages were filtered away. Alas.
When I disconnect the 120Ohm resistor I can see in PCAN-View error messaging coming in, with data as: "Other Error, Rx, Tolerate dominant bits, RxErr=128, TxErr=0". Since I receive this in PCAN-View I figured I should also somehow get these messages in python.

When adding some print values in _recv_internal, I can see that every second the result is first `PCAN_ERROR_BUSLIGHT | PCAN_ERROR_BUSHEAVY` and then 2 or 3 times it `HAS_EVENTS` and isn't equal to `WAIT_OBJECT_0`.
I just thought that when PCAN-View gets an error message (in my current setup each 30ms), this module would also receive it in the `_recv_internal` function

christiansandberg, 2019-10-16:
Maybe there is some initialization or parameter setting that needs to be done for error frames to be enabled?

tamenol, 2019-10-16:
Maybe, that's why I come here :) As you can see, I disable any filter by `self.bus.set_filters()` so every message should be matched

christiansandberg, 2019-10-16:
python-can mostly relies on the users to provide patches and new interfaces as the maintainers usually don’t have access to the hardware. It would be best if you could look up the API documentation and try to provide a fix yourself.

tamenol, 2019-10-21:
Seems like I have a solution. I forked the develop branch and commited my change. Have to do some additional tests before I'll do a push request here.

Also, I use a bit of code to check if windows has the pcan driver. Is this also something worth adding to this module? I'd add it in basic.py, in the `__init__` function after checking that the dll is present.

`
        key_exists = True
        aReg = winreg.ConnectRegistry(None, winreg.HKEY_LOCAL_MACHINE)

        try:
            aKey = winreg.OpenKey(aReg, r"SOFTWARE\PEAK-System\PEAK-Drivers")
            winreg.CloseKey(aKey)
        except WindowsError:
            key_exists = False

        winreg.CloseKey(aReg)
`

commit 69451507 (2019-11-07): Add support for error frames on PCAN (#711)

Enable reception of error frames on the PCAN interface.

Fixes #707
closed 2019-11-07 by karlding