hardbyte/python-can #709: python-can 3.3.2 does not provide the extra 'seeedstudio'

PonyZhou, 2019-10-13:
Hi, I tryed to intall python-can for my USB-CAN Analyzer and typed "pip3 install python-can[seeedstudio]" commands as the official website suggests, but get the feedback "python-can 3.3.2 does not provide the extra 'seeedstudio'  "

karlding, 2019-10-16:
@hardbyte the ReadTheDocs site for `3.3.2` still points to the [bad `3.3.2` release tag](https://github.com/hardbyte/python-can/tree/3.3.2), instead of the docs for the `3.3.2` release available on PyPI.

PonyZhou, 2019-10-16:
So u means " "pip3 install python-can[seeedstudio]"" commands is not availabe anymore.

karlding, 2019-10-16:
That new interface wasn't meant to be released until the `4.x.x` series. The original plan for the `3.x.x` releases was only for bug-fixes, so this wrong tag really threw things in a loop. Technically I think we should've revoked that version and bumped the minor version to avoid this sort of confusion...

To answer your question, the interface still exists, but it's currently only on the `develop` branch, and will be picked up in the next major version. I'm not sure when the next planned release is, but to unblock yourself, you can install the latest version from the Git repo (not generally recommended, proceed at your own risk).

If you're using `pip`, something along these lines should work.

```bash
pip install -e git+git://github.com/hardbyte/python-can.git@7bf39285967d785843919f12a4f69ba736181586#egg=python-can[seeedstudio]
```

PonyZhou, 2019-10-16:
I'll try to install it. Thanks for reply !

felixdivo, 2021-04-15:
This seems to have worked. Please comment if there are still problems.

closed 2021-04-15 by felixdivo