hardbyte/python-can #712: PCAN filter slips up sometimes

tamenol, 2019-10-23:
I'm using the `bus.set_filters()` function to set a filter so I only receive messages with an TX_SDO id. Using the filter `{"can_id":0x580, "can_mask":0xF80}`. However, I've noticed that sometimes, the TX_PDO id also gets throught the filter wich messes up my polling after id's function.

```
import asyncio
import can
import logging

class PCAN(object):

    RX_SDO = 0x600
    TX_SDO = 0x580
    RX_PDO = 0x200
    TX_PDO = 0x180

    TX_mask = 0xF80

    TX_SDO_filter = {"can_id": 0x580,  # TX_SDO,
                     "can_mask": TX_mask}

    TX_PDO_filter = {"can_id": 0x180,  # TX_PDO,
                     "can_mask": TX_mask}

    ids = set([])

    def _extract_id(self, msg):
        print(msg)
        print("{} == {}".format(msg.arbitration_id & 0xF80,
                                0x580 & 0xF80))
        self.ids.add(msg.arbitration_id - self.TX_SDO)

    def __init__(self, channel="PCAN_USBBUS1"):

        self.logger = logging.getLogger("CAN")
        self.logger.info("Initializing CANbus")

        can.set_logging_level("info")

        self.bus = can.Bus(channel=channel, bustype="pcan")
        self.listener = []
        self.loop = asyncio.get_event_loop()
        self.notifier = can.Notifier(self.bus, self.listener, loop=self.loop)

    async def poll_id(self, canid):

        data = [0x40, 0x00, 0x10, 0x00]
        msg = can.Message(arbitration_id=self.RX_SDO + canid,
                          data=data,
                          is_extended_id=False)

        self.bus.send(msg)
        await asyncio.sleep(0.05)

    async def poll_ids(self):

        self.logger.info("Polling ids")
        self.ids = set([])
        self.bus.set_filters([self.TX_SDO_filter])
        self.notifier.add_listener(self._extract_id)
        res_ids = await asyncio.gather(*(self.poll_id(canid) for canid in range(128)))
        self.notifier.remove_listener(self._extract_id)
        self.logger.info("Found ids: {}".format(self.ids))
        self.bus.set_filters()

    def cleanup(self):
        """
        A function to clean everything up.
        """
        self.notifier.stop()
        self.bus.shutdown()
        self.logger.info("Stopped notifier; Bus shutdown")

if __name__ == "__main__":

    try:
        loop = asyncio.get_event_loop()
        cn = PCAN()
        loop.run_until_complete(cn.poll_ids())
        print(cn.ids)
    except KeyboardInterrupt:
        pass

    loop.close()
    cn.cleanup()

```

Output:

> C:\repo\loki-Hg\tools\productiondiagnostics>python test.py
> Timestamp:   413575.599074        ID: 05b2    S                DLC:  8    43 00 10 00 00 00 00 00
> 1408 == 1408
> Timestamp:   413575.610082        ID: 058a    S                DLC:  8    43 00 10 00 00 00 00 00
> 1408 == 1408
> {50, 10}
> 
> C:\repo\loki-Hg\tools\productiondiagnostics>python test.py
> Timestamp:   413576.607169        ID: 018a    S                DLC:  7    00 ff ff ff 00 00 00
> 384 == 1408
> Timestamp:   413576.628118        ID: 058a    S                DLC:  8    43 00 10 00 00 00 00 00
> 1408 == 1408
> Timestamp:   413576.641985        ID: 05b2    S                DLC:  8    43 00 10 00 00 00 00 00
> 1408 == 1408
> {-1014, 10, 50}
> 
> C:\repo\loki-Hg\tools\productiondiagnostics>python test.py
> Timestamp:   413577.560095        ID: 058a    S                DLC:  8    43 00 10 00 00 00 00 00
> 1408 == 1408
> Timestamp:   413577.573876        ID: 05b2    S                DLC:  8    43 00 10 00 00 00 00 00
> 1408 == 1408
> {10, 50}
> 

Notice the `-1014`, which is equal to `0x18A - 0x580`.

Does the `bus.set_filters` work for the PCAN interface or do I need to use the `FilterMessages(self, Channel, FromID, ToID, Mode)` function in `basic.py`? I don't find any usage of 'FilterMessages; in the rest of the code.

So my guess is that I need to write some function in `pcan.py` to use `FilterMessages`? Or is this a bug in `bus.py`?

karlding, 2019-10-25:
> Does the bus.set_filters work for the PCAN interface

`set_filters` simply assigns the filters to `self._filters`

```python
    def set_filters(self, filters: Optional[can.typechecking.CanFilters] = None):
        # ...
        self._filters = filters or None
        self._apply_filters(self._filters)

```

>  or do I need to use the FilterMessages(self, Channel, FromID, ToID, Mode) function in basic.py? I don't find any usage of 'FilterMessages; in the rest of the code.

The PCAN interface currently does not implement HW filtering at the driver-level. However, if a interface does not implement `_apply_filters` to take advantage of this in `_recv_internal`, `python-can` provides a default "fallback" filtering option outside of the driver via `_matches_filters`.

What you're describing seems to indicate that somehow `_matches_filters` is returning `True`, as the PCAN interface's `_recv_internal` never returns `True` for whether the messages were already filtered by HW.

```python
    def recv(self, timeout: Optional[float] = None) -> Optional[Message]:
        # ...
        start = time()
        time_left = timeout

        while True:

            # try to get a message
            msg, already_filtered = self._recv_internal(timeout=time_left)

            # return it, if it matches
            if msg and (already_filtered or self._matches_filters(msg)):
                LOG.log(self.RECV_LOGGING_LEVEL, "Received: %s", msg)
                return msg
        # ...

```

The issue you pointed out tracks the implementation of a theoretically more "optimized" filtering solution, as it currently does so outside of the driver. However, as the bug details, the semantics that the PCAN driver provides are slightly different from what `python-can` provides, so there is some work needed to flesh that design out. Regardless of whether HW filtering is used or not, filtering _should_ work (otherwise there's a bug somewhere).

tamenol, 2019-10-25:
Hey, thanks for your input. I won't have much time today to have a closer look at this. Will do in the course of next week. 
The hard part will be replicating this error I guess. haha

tamenol, 2019-11-21:
So, some time has passed and since then I rewrote my class. For polling the ID's on the bus I use now this code snippet:

```
    async def poll_ids(self):
        async def pollid(canid):

            msg = can.Message(
                arbitration_id=self.RX_SDO + canid,
                data=[0x40, 0x00, 0x10, 0x00],
                is_extended_id=False,
            )
            self.canbus.send(msg)

        self.logger.info("Polling ids")

        id_reader = can.AsyncBufferedReader()
        listen = [id_reader]
        self.canbus.set_filters([self.TX_SDO_filter])

        noti = can.Notifier(self.canbus, listen, loop=self.loop)

        current_ids = []

        for i in range(128):
            await pollid(i)
            try:
                msg = await asyncio.wait_for(id_reader.get_message(), timeout=0.05)
                current_ids.append(msg.arbitration_id - self.TX_SDO)
            except:
                pass
            if i % 16 == 0:
                self.logger.info("Polled {}/128 id's".format(i))
        self.logger.info("Polled 128/128 id's")

        noti.stop()
        self.canbus.set_filters([])

        self.logger.info("Found these ID's: {}".format(current_ids))

        current_ids.sort()
        return current_ids
```

I did not reproduce the slip ups of the filter. I started using local readers/notifiers and stopping them when not needed anymore.
I have honestly no idea what I did right this time, but I guess my issue in the opening post is due to my programming skills rather than a bug in the python-can module.

hardbyte, 2019-11-21:
I'm glad to hear that, please reopen the issue if you catch it misbehaving!

tamenol, 2020-02-20:
So, I caught it misbehaving again. When filtering for SDO messages. it could be that a PDO message gets through at the start. If it's coming on the `right` time. 
I'm not sure if it's my code or a fault in the python-can. Notice in the code I first set the filter, afterwards I add the listener to the notifier.

```
import asyncio
import can
import logging
from typing import List

RX_SDO = 0x600
TX_SDO = 0x580
RX_PDO = 0x200
TX_PDO = 0x180

TX_mask = 0xF80

TX_SDO_filter = {"can_id": TX_SDO, "can_mask": TX_mask}
TX_PDO_filter = {"can_id": TX_PDO, "can_mask": TX_mask}

class MaPCANBUS(object):

    devices = {}

    def _log_recv_msg(self, msg: can.Message) -> None:
        self.logger.debug(f"RECEIVED message\n\t{msg}")

    def __init__(self, channel: str = "PCAN_USBBUS1", bustype: str = "pcan"):

        self.logger = logging.getLogger("CAN")
        self.logger.info("Initializing PCANbus")
        can.set_logging_level("info")

        self.canbus = can.Bus(channel=channel, bustype=bustype)
        self.notifier = can.Notifier(self.canbus, [], loop=asyncio.get_event_loop())

        # if self.logger.getEffectiveLevel() == logging.DEBUG:
        #     self.notifier.add_listener(self._log_recv_msg)

    async def _send_message(
        self, device_id: int, data: List[int], wait: float = 0.001
    ) -> None:
        msg = can.Message(arbitration_id=device_id, data=data, is_extended_id=False)
        self.logger.debug(f"SEND message\n\t{msg}")
        self.canbus.send(msg)
        # sleep is needed to not spam the microcontroller with messages, otherwise it's possible some msgs will be skipped. Depending on what's asked of the microcontroller.
        # -i think
        await asyncio.sleep(wait)

    async def get_device_ids(self) -> None:
        def _add_id_to_list(msg: can.Message) -> None:
            self.logger.debug(f"RECEIVED message\n\t{msg}")
            self.devices[msg.arbitration_id - TX_SDO] = ""

        async def _run() -> None:
            for i in range(128):
                await self._send_message(RX_SDO + i, [0x40, 0x00, 0x10, 0x00])

        self.logger.info("Polling ids")

        self.canbus.set_filters([TX_SDO_filter])
        self.notifier.add_listener(_add_id_to_list)

        await asyncio.get_event_loop().create_task(_run())

        self.notifier.remove_listener(_add_id_to_list)
        self.canbus.set_filters([])

        self.logger.info(f"ids found: {self.devices.keys()}")

    async def get_device_names(self, refresh_ids: bool = False) -> None:
        def _add_name_to_list(msg: can.Message) -> None:
            self.devices[msg.arbitration_id - TX_SDO] = str(
                msg.data.decode().strip().strip("\x00")
            )

        async def _run() -> None:
            for i in self.devices.keys():
                await self._send_message(RX_SDO + i, [0x60])

        self.logger.info("Getting names")

        if refresh_ids or not self.devices:
            self.logger.info("first refreshing ids")
            await self.get_device_ids()

        self.canbus.set_filters([TX_SDO_filter])
        self.notifier.add_listener(_add_name_to_list)

        await asyncio.get_event_loop().create_task(_run())

        self.notifier.remove_listener(_add_name_to_list)
        self.canbus.set_filters([])

        self.logger.info(f"devices on the bus: {self.devices}")
```

And running: (in my program it's a bit complicater than this, since I'm running this in a PyQt5 application)

```
logging.basicConfig(level=logging.DEBUG)
loop = asyncio.get_event_loop()
test = MaPCANBUS()
loop.run_until_complete(test.get_device_ids())
```

And the logs: (this shows the 1 PDO received message (the first one))
```
DEBUG:asyncio:Using proactor: IocpProactor
INFO:SM206_window:Reading parameters from the SM206
INFO:CAN:Initializing PCANbus
INFO:CAN:Polling ids
DEBUG:CAN:RECEIVED message
	Timestamp:    88232.200158        ID: 018a    S                DLC:  7    00 ff ff ff 00 00 00
DEBUG:CAN:SEND message
	Timestamp:        0.000000        ID: 0600    S                DLC:  4    40 00 10 00
DEBUG:CAN:SEND message
	Timestamp:        0.000000        ID: 0601    S                DLC:  4    40 00 10 00
DEBUG:CAN:SEND message
	Timestamp:        0.000000        ID: 0602    S                DLC:  4    40 00 10 00
DEBUG:CAN:SEND message
	Timestamp:        0.000000        ID: 0603    S                DLC:  4    40 00 10 00
DEBUG:CAN:SEND message
	Timestamp:        0.000000        ID: 0604    S                DLC:  4    40 00 10 00
DEBUG:CAN:SEND message
	Timestamp:        0.000000        ID: 0605    S                DLC:  4    40 00 10 00
DEBUG:CAN:SEND message
	Timestamp:        0.000000        ID: 0606    S                DLC:  4    40 00 10 00
DEBUG:CAN:SEND message
	Timestamp:        0.000000        ID: 0607    S                DLC:  4    40 00 10 00
DEBUG:CAN:SEND message
	Timestamp:        0.000000        ID: 0608    S                DLC:  4    40 00 10 00
DEBUG:CAN:SEND message
	Timestamp:        0.000000        ID: 0609    S                DLC:  4    40 00 10 00
DEBUG:CAN:SEND message
	Timestamp:        0.000000        ID: 060a    S                DLC:  4    40 00 10 00
DEBUG:CAN:RECEIVED message
	Timestamp:    88232.242812        ID: 058a    S                DLC:  8    43 00 10 00 00 00 00 00
DEBUG:CAN:SEND message
	Timestamp:        0.000000        ID: 060b    S                DLC:  4    40 00 10 00
DEBUG:CAN:SEND message
	Timestamp:        0.000000        ID: 060c    S                DLC:  4    40 00 10 00
# ....
```

christiansandberg, 2020-02-20:
I’m not sure but changing the hardware filters on the fly could perhaps lead to subtle delays and previously buffered messages coming in anyway or something. Unless you have strong performance reasons to dynamically changing filtering on hardware level you should probably just filter them yourself in your receiving function instead.

tamenol, 2020-02-21:
Thanks for getting back at me. I see. I'll do that for now and see if there are performance issues. For now I'm using my application on a bus with 1 device that spits out his state with PDO messages at 300Hz. 
Ultimately my class will also get used on busy buses, nearing the limit of 500kbit/s, plotting the bit throughput (PDO messages) while talking with the devices with SDO messages (to see what devices are on the bus at that moment)

Will do some more testing.

closed 2020-02-27 by tamenol