hardbyte/python-can #720: send_periodic with SocketCan fails to stop when several tasks are running

random-username1, 2019-11-05:
Hi,

Looks like trying to stop several send_periodic tasks at the same time fails with SocketCan on Raspberry Pi.
```python3
task = bus.send_periodic(msg, period)
task2 = bus.send_periodic(msg2, period2)
task.stop()
task2.stop()
```
Here is the error message:
```
Traceback (most recent call last):
  File "/home/pi/venv/lib/python3-6/site-packages/can/interfaces/socketcan/socketcan.py", line 280, in send_bcm
    return bcm_socket.send(data)
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception accurred:

Traceback (most recent call last):
  File "test.py", line 9, in <module>
    task2.stop()
  File "/home/pi/venv/lib/python3-6/site-packages/can/interfaces/can/bus.py", line 218, in wrapped_stop_method
    original_stop_method()
  File "/home/pi/venv/lib/python3-6/site-packages/can/interfaces/socketcan/socketcan.py", line 365, in stop
    send_bcm(self.bcm_socket, stopframe)
  File "/home/pi/venv/lib/python3-6/site-packages/can/interfaces/socketcan/socketcan.py", line 285 in send_bcm
    raise can.CanError(base + "You are probably referring to a non-existing frame.")
can.CanError: Couldn't send CAN BCM frame. OS Error 22: Invalid argument
You are probably referring to a non-existing frame.
```

I'm using python-can 3.3.2 on a 32 bits Raspbian stretch (v9.4)

Here is the output of uname -a:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l GNU/Linux
```

karlding, 2019-11-06:
Out of curiosity, do your messages have the same arbitration ID?

random-username1, 2019-11-06:
All messages have different datas and periods but do have the same arbitration ID.

This may help:

- In `bus.py`, the `self._periodic_tasks` list of the **BusABC** class does contain all the tasks but it looks like only the last one is considered.
- I've also notised that trying to stop a task which has already been stopped will result in the same error.
- Also, this issue is probably linked with this one #721 

I'll keep searching and tell you if I find more

karlding, 2019-11-06:
I believe this was fixed by #604 and #605. Can you check if you still see this on `master`?

random-username1, 2019-11-07:
I forked the repository yesterday and looked at the code and it seems that this has been fixed but not released yet.
Having several messages with the same arbitration ID now raises a `ValueError` since I belive the socket doesn't allows it.
The best solution to send several CAN messages with the same arbitration ID at different periods appears to be a threarded loop in which a message is sent once at every iteration.

closed 2020-07-18 by hardbyte