hardbyte/python-can #721: Calling send_periodic several times doesn't work with SocketCan

random-username1, 2019-11-06:
Hi,
```python3
task = bus.send_periodic(msg, period)
task2 = bus.send_periodic(msg2, period2)
```
After launching several send_periodic tasks, with SocketCan, only the last one appears to be working although the same script works fine with PeakCan on another device.

Plugging this other device on the CAN Bus to intercept frames confirms that the last send_periodic task to be started is the only one working.

I'm using python-can 3.3.2 on a 32 bits Raspbian stretch (v9.4)

Here is the output of uname -a:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l GNU/Linux
```

christiansandberg, 2019-11-06:
Why do you need to send multiple messages with the same CAN-ID? SocketCAN does not support that and should generally never be needed. If it is a multiplexed message then you can provide a list of messages to `send_periodic` instead.

```python
bus.send_periodic([msg, msg2], period)
```

random-username1, 2019-11-06:
In my case, several CAN messages have the same CAN-ID and the first byte of data is used to 
differentiate them.
These messages have to be sent at different periods so I cannot give a list of messages to the `send_periodic()` method.

Then, wouldn't it be better to just call `bus.send()` in a threaded loop (like it's been done with the `class ThreadBasedCyclicSendTask:` of  python-can's `broadcastmanager.py` file) ?

Also I this this issue is related to #720

christiansandberg, 2019-11-06:
If you really need to do that then you should probably implement the periodic sending yourself like you say or use ThreadBasedCyclicSendTask directly somehow. The timing will be worse since it is done in Python instead of directly by the kernel so it is best avoided if possible.

karlding, 2019-11-08:
> Why do you need to send multiple messages with the same CAN-ID? SocketCAN does not support that and should generally never be needed.

To clarify, IIRC I believe that the restriction that all the CAN messages controlled by the BCM socket have the same Arbitration ID is one that `python-can` chooses to enforce, rather than one that the Linux kernel (SocketCAN interface) has. The kernel driver code basically uses the CAN Arbitration ID to identify the BCM TX Task, but there's no requirement that the Arbitration IDs of all the messages within the Task are the same (the values are copied straight from the array of messages). The fact that they call the attribute `can_id` is rather misleading, since it's more of a Task Identifier.

I think we chose to only allow the same Arbitration ID to simplify the logic on `python-can`'s side, since we're taking the Task Identifier from the first message's Arbitration ID. But there's nothing stopping us from generating a unique Task ID and setting it on the Task object instead of taking the CAN ID from the Messages..

christiansandberg, 2019-11-10:
It’s probably because the SocketCAN documentation seem to assume that the CAN ID should be used to identify the task. It does not say if other IDs will work or not. It would be great if you could try using a different can_id for the task. Also if it has to be unique per BCM socket or the whole bus.

karlding, 2019-11-13:
> It does not say if other IDs will work or not. It would be great if you could try using a different can_id for the task.

The kernel driver code doesn't associate the Cyclic Task ID with the CAN ID, other than sharing the same `canid_t` type.

> Also if it has to be unique per BCM socket or the whole bus.

Looking at the [Linux driver code](https://elixir.bootlin.com/linux/latest/source/net/can/bcm.c), it seems like the BCM Operations are per-BCM socket, rather than per-interface.

```c
static int bcm_tx_setup(struct bcm_msg_head *msg_head, struct msghdr *msg,
			int ifindex, struct sock *sk)
{
	struct bcm_sock *bo = bcm_sk(sk);
	struct bcm_op *op;
	struct canfd_frame *cf;
	unsigned int i;
	int err;

	/* we need a real device to send frames */
	if (!ifindex)
		return -ENODEV;

	/* check nframes boundaries - we need at least one CAN frame */
	if (msg_head->nframes < 1 || msg_head->nframes > MAX_NFRAMES)
		return -EINVAL;

	/* check timeval limitations */
	if ((msg_head->flags & SETTIMER) && bcm_is_invalid_tv(msg_head))
		return -EINVAL;

	/* check the given can_id */
	op = bcm_find_op(&bo->tx_ops, msg_head, ifindex);
	if (op) {
// ...
```

Here's a quick test program to verify.

```c
#include <linux/can.h>
#include <linux/can/bcm.h>
#include <net/if.h>
#include <stdbool.h>
#include <string.h>
#include <stdlib.h>
#include <sys/ioctl.h>
#include <sys/socket.h>
#include <sys/time.h>
#include <unistd.h>

#define TEST_FIRST_FRAME_DATA 0x11U
#define TEST_SECOND_FRAME_DATA 0x22U
#define TEST_THIRD_FRAME_DATA 0x33U

#define TEST_FIRST_FRAME_DLC 1U
#define TEST_SECOND_FRAME_DLC 1U
#define TEST_THIRD_FRAME_DLC 1U

#define TEST_NUM_FRAMES 3U

#define TEST_CAN_INTERFACE_NAME "vcan0"

int main(int argc, char *argv[]) {
  int fd = socket(PF_CAN, SOCK_DGRAM, CAN_BCM);
  struct ifreq ifr = { 0 };
  struct sockaddr_can addr = { 0 };

  struct bcm_timeval initial_period = {
    .tv_sec = 0U,
    .tv_usec = 0U,
  };
  struct bcm_timeval subsequent_period = {
    .tv_sec = 1U,
    .tv_usec = 0U,
  };

  struct {
    struct bcm_msg_head msg_head;
    struct can_frame frame[TEST_NUM_FRAMES];
  } msg;

  strcpy(ifr.ifr_name, TEST_CAN_INTERFACE_NAME);
  ioctl(fd, SIOCGIFINDEX, &ifr);

  addr.can_family = AF_CAN;
  addr.can_ifindex = ifr.ifr_ifindex;

  msg.msg_head.opcode = TX_SETUP;
  msg.msg_head.can_id = (canid_t)atoi(argv[1]);
  msg.msg_head.flags = SETTIMER | STARTTIMER;
  msg.msg_head.nframes = TEST_NUM_FRAMES;
  msg.msg_head.ival1 = initial_period;
  msg.msg_head.ival2 = subsequent_period;
  msg.msg_head.count = 0U;

  msg.frame[0].can_id = (canid_t)(atoi(argv[2]) + 0);
  msg.frame[0].can_dlc = TEST_FIRST_FRAME_DLC;
  msg.frame[0].data[0] = TEST_FIRST_FRAME_DATA;

  msg.frame[1].can_id = (canid_t)(atoi(argv[2]) + 1);
  msg.frame[1].can_dlc = TEST_FIRST_FRAME_DLC;
  msg.frame[1].data[0] = TEST_SECOND_FRAME_DATA;

  msg.frame[2].can_id = (canid_t)(atoi(argv[2]) + 2);
  msg.frame[2].can_dlc = TEST_THIRD_FRAME_DLC;
  msg.frame[2].data[0] = TEST_THIRD_FRAME_DATA;

  connect(fd, (struct sockaddr *)&addr, sizeof(addr));
  write(fd, &msg, sizeof(msg));

  while (true) {
    /* Spin */
  }

  close(fd);

  return 0;
}

```

Running like so (note that here, the Cyclic Task is started with `can_id = 1`, despite the Arbitration IDs being `0x401`, `0x402`, `0x403`):

```
 $ ./test 1 1025
  vcan0  401   [1]  11
  vcan0  402   [1]  22
  vcan0  403   [1]  33
  vcan0  401   [1]  11
  vcan0  402   [1]  22
  vcan0  403   [1]  33
 ...
```

And then running two applications at the same time (note that this results in different BCM sockets being opened on the same SocketCAN network interface)

```
 $ ./test 1 1025
 $ ./test 1 769
  vcan0  401   [1]  11   <-- ./test 1 1025
  vcan0  303   [1]  33   <-- ./test 1 769
  vcan0  402   [1]  22   <-- ./test 1 1025
  vcan0  301   [1]  11   <-- ./test 1 769
  vcan0  403   [1]  33   <-- ./test 1 1025
  vcan0  302   [1]  22   <-- ./test 1 769
  vcan0  401   [1]  11   <-- ./test 1 1025
  vcan0  303   [1]  33   <-- ./test 1 769
  vcan0  402   [1]  22   <-- ./test 1 1025
  vcan0  301   [1]  11   <-- ./test 1 769
  vcan0  403   [1]  33   <-- ./test 1 1025
  vcan0  302   [1]  22   <-- ./test 1 769
```

If we're okay with making this change, I can put up a patch to switch the behaviour of the SocketCAN interface such that we:

1. Keep a global counter of the next available Task ID (or at least track the currently active Task IDs)
2. Store the Cyclic Task ID on the `Task` object we return
3. Remove the restriction that all `Messages` need the same Arbitration ID

tysonite, 2019-12-05:
@karlding That is a very nice improvement. I would love to have it backported to 3.3.x version.

tysonite, 2020-03-03:
Would you please review the PR above? It removes restriction on uniqness of arbitration ID.

commit 2d672b34 (2020-04-09): Relax restriction of arbitration ID uniqueness for SocketCAN (#785)

Socketcan bcm now supports sending periodic messages with the same arbitration ID.
Updates documentation and tests for socketcan interface.

Closes #721
closed 2020-04-09 by hardbyte