hardbyte/python-can #726: canalyst-ii libcontrolcan.so?

expntly, 2019-11-14:
Sorry for posting this here but I can't figure out how to use the canalystii interface on Linux with python-can.

`can/interfaces/canalystii.py` wants to open `libcontrolcan.so`, which I don't have on my system and I have no idea where it's coming from after searching online.  I was expecting this was a library provided by the manufacturer / ZHIYUAN but I can't find it on their website (mostly Windows files).

A pointer or a note in the doc would help thanks!

https://python-can.readthedocs.io/en/master/interfaces/canalystii.html

altrunner, 2020-02-15:
Just in case it is still actual - I downloaded a copy of libcontrolcan.so from here https://github.com/AbangLZU/can_control/tree/master/can_control_node/lib

jxltom, 2020-02-27:
The docs should be updated to reflect the fact that these files should be provided by user themselves.

kdschlosser, 2020-11-14:
The problem with supplying the files is that there are clones of the canalyst device. while you may think that you have purchased a device that is made by Zhiyuan Electronics you may have gotten one from a manufacturer that has cloned the device. At a minimum there are changes to the libraries API that deal with how to attach to the device. 

A CANalyst-II purchased from Amazon is not going to be made by the manufacturer mentioned above. Instead it is made by Chuangxin Technology and the drivers/library for Linux can be downloaded from this link
http://24869997.s21d-24.faiusrd.com/0/ABUIABAAGAAgpqj__AUomOuchAM?f=CAN%E5%88%86%E6%9E%90%E4%BB%AA%E8%B5%84%E6%96%9920200702_Linux.7z&v=1595905068

anand-bala, 2021-07-16:
I realize this issue is old, but would it be possible to change `CANalystII = CDLL("./libcontrolcan.so")` in [line 65](https://github.com/hardbyte/python-can/blob/5b63bb28bd849dcf273cd648e32db53e21daf9d9/can/interfaces/canalystii.py#L65) to `CANalystII = CDLL("libcontrolcan.so")"?

Removing the "./" in that allows `ctypes` to search `LD_LIBRARY_PATH` or use something like `ldconfig` to find the library, as opposed to hard coding it to search only in the current working directory.

Cherish-forever, 2021-08-30:
@altrunner  
why no cpoy libcontrolcan.so to python-can/can/interfaces ?
and use relative script file path.

commit 18f34621 (2021-11-14): canalystii: Switch from binary library to pure Python driver

- Project 'canalystii' hosted at https://github.com/projectgus/python-canalystii

Compared to previous implementation:

- No more binary library, can support MacOS as well as Windows and Linux
- Can receive on all enabled channels (previous implementation received on one channel only)
- Performance is slightly slower but still easily possible to receive at max
  message rate on both channels
- send timeout is now based on CAN layer send not USB layer send
- User configurable software rx message queue size
- Adds tests (mocking at th [... cut]