hardbyte/python-can #728: NameError: name 'UcanInitHardwareEx' is not defined

Mango-kid, 2019-11-26:
This error is in the systec interface when initalizing the bus. 

`bus = can.interface.Bus(bustype='systec', channel='0', bitrate=500000)`

karlding, 2020-04-23:
Is this still occurring? If so, can you provide more information (`python-can` library version, Python version, whether you installed the SYS TEC drivers, etc.)?

`python-can` requires SYS TEC CAN device drivers and the `usbcan32.dll` (or `usbcan64.dll` for 64-bit).

ryleymcc, 2021-03-12:
I'm getting this error. I am using python 3.9.2 on windows 10. I have downloaded and run the driver setup and i can see the .dll the folder
[image]
 I do not have my systec USBCAN plugged in.

felixdivo, 2022-02-06:
Loading of the ucan DLL probably failed and got unnoticed. It is probably fixed by #1158, see [this](https://github.com/hardbyte/python-can/pull/1158/files#diff-24430eec1b6a63fd7a7938badd4d7665fc44f31ccf7a465b2c375e3286dcab11R331-R335). If the problem persists, please reopen or comment.

closed 2022-02-06 by felixdivo