hardbyte/python-can #730: SocketCAN J1939 integration

olerem, 2019-11-30:
Hi,

I'm kernel developer and spend last two years on preparing of J1939 kernel stack for mainline. Now it is happened and we have this stack in kernel 5.4. The UAPI is stable now and can be safely used.

Do you have plans to integrate it in to python-can? I can answer you questions regarding kernel implementation, but currently I do not have enough experience in python. So, it will be great if some one can help to integrate it.

christiansandberg, 2019-11-30:
That’s awesome news! This library is however deliberately for raw CAN only that other applications or packages can build higher level protocols upon. See https://github.com/pylessard/python-can-isotp as an example. It provides a single interface for both SocketCAN ISO-TP and and a Python implementation for other OS and interfaces. Something similar for J1939 would be very useful.

closed 2019-12-16 by christiansandberg

karlding, 2020-04-30:
FYI @olerem, my CPython patch that adds support for `CAN_J1939` sockets in the Python standard library landed today.

It should be available in the `socket` module in Python 3.9+ (in releases after and including beta 1), or at `HEAD` if you would like to build your own release to play around with before the actual beta 1 release (May 18, 2020 per the schedule in PEP 596).

**Note**: I don't think SocketCAN tests that require an interface (virtual or physical) actually run on the Python BuildBot cluster, so please do use it and report any bugs, so they can be addressed prior to the final 3.9 release version on October. 🙂

milhead2, 2021-06-29:
I have kept alive a python-j1939 library but nowhere near the support level that python-can has.  At https://github.com/milhead2/python-j1939,,,  We use it to simulate ECU's under a robotframework environment.  It certanly is a potential refactor to use the kernel j1939 support but probably not until we get all our development and test workstarions updated to >5.4 kernel.  What is today a pretty ugly and slowish j1939 library could become a very thin high-performance layer..  I'm looking forward to that.

olerem, 2021-06-29:
please note, the kernel is using socket error queue to notify about transfer status updates. The TX variant is implemented and mainline. The RX is currently under development.