hardbyte/python-can #732: XL_ERR_NOT_IMPLEMENTED when calling xlGetSyncTime on VN8900

danielhrisca, 2019-12-02:
Hello,

I'm trying to use a VN8900 device but get this error during the __init__ https://github.com/hardbyte/python-can/blob/release-3.3.2/can/interfaces/vector/canlib.py#L209 (using version 3.3.2). 

I have multiple VN devices connected (including a VN7600), but I don't use the serial number, I just assign a custom applciation name in the vector hardware configuration utility.

christiansandberg, 2019-12-02:
Could be that some hardware does not support reading the current timestamp. In that case we could just fall back to relative timestamps.

danielhrisca, 2019-12-02:
It seems that there is a dedicated functionf or the VN8900 family (see 3.2.22 xlGetChannelTime
in https://assets.vector.com/cms/content/products/XL_Driver_Library/Docs/XL_Driver_Library_Manual_EN.pdf )

I will try to see if this fixes the problem

danielhrisca, 2019-12-02:
@christiansandberg 

Using the fix from the PR I was able to access the bus and read messages.

commit f342c1c6 (2019-12-02): fixes #732: add support for VN8900 xlGetChannelTime function (#733)
commit 966a5e4d (2021-01-18): Windows time accuracy (#936)

* fixes #732: add support for VN8900 xlGetChannelTime function
* improve time stamp accuracy of the CAN messages for Kvaser and Vector buses on the Windows platform. (see #934)
On Windows the value returned by time.time is refreshed at best every 0.5ms, and on some PCs the default refresh rate is every 10ms. The idea is to get the value of time.perf_counter at the moment when time.time is refreshed and use this as a reference point.
* check the time.time resolution before using the new time correlation function
commit 2e24af08 (2022-01-10): Fix vector get application config (#977)

* fixes #732: add support for VN8900 xlGetChannelTime function

* add another level of try/except according to the review

* format using black

* the application channels needs to be provided as a ctypes variable

* fix mock in test_vector.py
closed 2019-12-02 by christiansandberg