hardbyte/python-can #735: Issue in recognising the pcan with python code

queen-king, 2019-12-03:


tamenol, 2019-12-03:
Can you try this:

`bus = can.Bus(channel='PCAN_USBBUS1', bustype='pcan', bitrate=250000)`

queen-king, 2019-12-04:
It's giving attribute error.
It says "module can has no attribute bus".

tamenol, 2019-12-04:
It's `Bus` with a capital letter. Maybe that's it? But I get a different error when I try that. What version are you running?

```
Python 3.6.8 (tags/v3.6.8:3c6b436a57, Dec 24 2018, 00:16:47) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import can
>>> bus = can.Bus(channel="PCAN_USBBUS1", bustype="pcan")
>>> bus
<can.interfaces.pcan.pcan.PcanBus object at 0x000002367FE2C358>
>>> print(bus)
PCAN_USBBUS1
```

```
~ $ pip show python-can
Name: python-can
Version: 3.2.0
```

queen-king, 2019-12-04:
I am using 3.8 version. i have used capital 'B' in the code. It still says attribute error.

queen-king, 2019-12-04:
Python 3.7.4 (tags/v3.7.4:e09359112e, Jul  8 2019, 19:29:22) [MSC v.1916 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import can
>>> bus1 = can.Bus(channel="PCAN_USBBUS1", bustype="pcan")

This is the error we are getting for the same code.

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'can' has no attribute 'Bus'
>>>

tamenol, 2019-12-04:
You do that in a clean (new) python interpreter shell? No other files called `can.py` in that directory? What is the output of `pip show python-can`?

It's weird. I can even do `b = can.interface.Bus(bustype="pcan")` in a new shell and it runs without errors.

queen-king, 2019-12-05:
There was a dll path missing in the basic.py. After the path was changed,we got the output. Thank you

natek512, 2020-01-27:
Hi! Can you tell me what your exact fix was? I am having the same issue and I am not really sure where you are setting this dll path. Thanks!

felixdivo, 2020-04-29:
If the problem persists, please open a new issue where you for example state on what platform you are working.

closed 2020-04-29 by felixdivo

tarunkumar3096, 2021-08-31:
I'm getting same error "AttributeError: module 'can' has no attribute 'Bus'"
@queen97king2019 Can you tell exact fix to resolve this error

felixdivo, 2021-09-05:
It's probably importing some other module that's also named can. Please inspect `can.__file__` or similar attributes to find out ehere it is imported from.