hardbyte/python-can #737: Absence of can.Message.__eq__

tysonite, 2019-12-05:
I noticed that ` can.Message.__eq__` has been removed in [this](https://github.com/hardbyte/python-can/commit/c37420d23e17f48067e7cc2de4778a88c743061e#diff-7080e7643f241026215c54f04aa5b7eaL132) revision. Is it really expected?

It was useful in verification of mocked calls in tests, as well as just comparing diffrerent messages. I understand that `equals()` provides the comparison mechanism now, but it looks to me that built-in equality function gives better flexibility, where not necessary to manage comparsion of timestamps.

christiansandberg, 2019-12-05:
The main issue is that there is no universal definition of when two messages are equal. Especially if the messages are different in time or appear on different channels.

tysonite, 2019-12-05:
Can those restrictions be relaxed while comparing two messages? The same issue applies to `equals()`, right? But it is still there.

christiansandberg, 2019-12-05:
With a method you can provide arguments controlling the behavior so it will be more flexible depending on your specific needs. You can now choose if you want to compare timestamps or not. The implementation of __eq__ will always be fixed and never be able to satisfy all use cases.

tysonite, 2019-12-05:
I do agree with what you are saying. On the other hand, invocation of a method requires specifying arguments if behaviour has to be changed. If you are fine with default behaviour, arguments are just omitted as they have default values.

I am wondering if two methods can co-exist. The **eq** calls **equals** by default without special arguments, but if anyone needs specialization, he calls **equals** specifically with a set of required arguments.

tysonite, 2019-12-05:
BTW, please feel free to close this issue if the project direction has been already confirmed in regards of that.

christiansandberg, 2019-12-07:
@felixdivo what do you think?

felixdivo, 2019-12-07:
Hey :) It's been a while ...

On a technical level, two methods could co-exist and call each other, sure. I think it would be okay (and I personally think also a good idea) to provide `__eq__` like that. It's the usual discussion on simple but possibly misleading vs. explicit but somewhat complicated. However, it is again a breaking change! (Although probably many will not have a problem with this change)

We should check that the documentation in the code explains how comparison works. And there should be some rationale on this in the "user guide" documentation as well.

felixdivo, 2020-04-29:
Done by #747.

closed 2020-04-29 by felixdivo