hardbyte/python-can #738: Incorrect pypi wheel which might destroy other packages

chrisoro, 2019-12-06:
When installing your released pypi packaged it will install the following taken from RECORD:

> ../../Scripts/__pycache__/can_logger.cpython-37.pyc,,
> ../../Scripts/__pycache__/can_player.cpython-37.pyc,,
> ../../Scripts/__pycache__/can_viewer.cpython-37.pyc,,
> ../../Scripts/can_logger.py,sha256=mLcSR0czHD0kWfA-zRXy_VT64OsiJewRfvtcQB1lL38,166
> ../../Scripts/can_player.py,sha256=PW1zDei0tTGKpqRQLw1ZSyiMyuZM2R7p9Iwzks2EUYo,166
> ../../Scripts/can_viewer.py,sha256=2iiEVNNM6WBhiBOLwHWPU9qn4Sq0R38AJxTBtjjfZQA,166
> can/CAN.py,sha256=gm1FzhF3QVj_5W7HQkPrhIkZrqAb13vi9S6Rxicuuh8,941
> can/__init__.py,sha256=UaX_PSKHNZbHRg9visUW5k1wWj0CbjlC3CCgpqoBONo,1227
> can/__pycache__/CAN.cpython-37.pyc,,
> can/__pycache__/__init__.cpython-37.pyc,,
> can/__pycache__/broadcastmanager.cpython-37.pyc,,
> can/__pycache__/bus.cpython-37.pyc,,
> can/__pycache__/ctypesutil.cpython-37.pyc,,
> can/__pycache__/interface.cpython-37.pyc,,
> can/__pycache__/listener.cpython-37.pyc,,
> can/__pycache__/logger.cpython-37.pyc,,
> can/__pycache__/message.cpython-37.pyc,,
> can/__pycache__/notifier.cpython-37.pyc,,
> can/__pycache__/player.cpython-37.pyc,,
> can/__pycache__/thread_safe_bus.cpython-37.pyc,,
> can/__pycache__/util.cpython-37.pyc,,
> can/__pycache__/viewer.cpython-37.pyc,,
> can/broadcastmanager.py,sha256=aK7ACLJqXEijHfAp6HFZxhlQRQAe0V_y-lo6Sh2RAqk,5184
> can/bus.py,sha256=zRnIsqI8u42DoIN5Gk9KK3LN4O7tDrbwEvcnI8_P27w,13594
> can/ctypesutil.py,sha256=0gXuSRHeUvliDZpgRAY4HwR9b58ZFM4RORqpBN_C4to,2826
> can/interface.py,sha256=W7G9_HxF9iODgHHpF4yT5GEK1ZNBdtvdMac98HHBmuc,6049
> can/interfaces/__init__.py,sha256=v9jv3MnHx50bmDfYF_73RF3fBbhwVVzJqRCdUTirnwg,1863
> can/interfaces/__pycache__/__init__.cpython-37.pyc,,
> can/interfaces/__pycache__/canalystii.cpython-37.pyc,,
> can/interfaces/__pycache__/iscan.cpython-37.pyc,,
> can/interfaces/__pycache__/nican.cpython-37.pyc,,
> can/interfaces/__pycache__/slcan.cpython-37.pyc,,
> can/interfaces/__pycache__/virtual.cpython-37.pyc,,
> can/interfaces/canalystii.py,sha256=26sM44vjeqhvueo8-k7gTN3O_J0_F9LYEFwuIBzsb14,5292
> can/interfaces/ics_neovi/__init__.py,sha256=L9JPBhdEszuLZF1PX8DSWTUkwsUU3TP2WVsSi0zzWNo,82
> can/interfaces/ics_neovi/__pycache__/__init__.cpython-37.pyc,,
> can/interfaces/ics_neovi/__pycache__/neovi_bus.cpython-37.pyc,,
> can/interfaces/ics_neovi/neovi_bus.py,sha256=NpXt4IifSGLzWHx4HjpPFY9tJeXxQLhjkHeG0He1_Ow,12375
> can/interfaces/iscan.py,sha256=cBaDWCl8WwW6DfOCRFBFhHwX2wE7DklBOPd-W0KDp2Y,5740
> can/interfaces/ixxat/__init__.py,sha256=9mPf6xj7g2e3VHq7QAS1_Ni3bvbb57JyMYusw8YsP-4,222
> can/interfaces/ixxat/__pycache__/__init__.cpython-37.pyc,,
> can/interfaces/ixxat/__pycache__/canlib.cpython-37.pyc,,
> can/interfaces/ixxat/__pycache__/constants.cpython-37.pyc,,
> can/interfaces/ixxat/__pycache__/exceptions.cpython-37.pyc,,
> can/interfaces/ixxat/__pycache__/structures.cpython-37.pyc,,
> can/interfaces/ixxat/canlib.py,sha256=Gra7biamkXIEylOsvkRLCrFQPmFvd5zmz2hHPI0rDUk,28099
> can/interfaces/ixxat/constants.py,sha256=_lT2lN6z6H0Tt2WdlNj5oAbQE441CmOZFlIShjEfqG8,4537
> can/interfaces/ixxat/exceptions.py,sha256=2qqzBT0Usr7IGmQOFfkPg3erjSgKX4T0GjNTRxFXSaY,726
> can/interfaces/ixxat/structures.py,sha256=5qJeo0aHFnkEHidIn-lW86kLzjSZi8A_sLiX9Du1tIM,4573
> can/interfaces/kvaser/__init__.py,sha256=ri6sk6zxt74cpch8dnM3i-jFYSteG_blgVWUgBiboG0,69
> can/interfaces/kvaser/__pycache__/__init__.cpython-37.pyc,,
> can/interfaces/kvaser/__pycache__/canlib.cpython-37.pyc,,
> can/interfaces/kvaser/__pycache__/constants.cpython-37.pyc,,
> can/interfaces/kvaser/__pycache__/structures.cpython-37.pyc,,
> can/interfaces/kvaser/canlib.py,sha256=V-n8okDdYSQmJNDmSshqcxOx5XzHH2w49Sp6DGXan3Y,25968
> can/interfaces/kvaser/constants.py,sha256=BAhuf9BOlzidGgJBKzaJc02U5-Bo_Uy9aKtkjsst2GM,6918
> can/interfaces/kvaser/structures.py,sha256=FaMFbdFtZVTxopvh0IOgYSWO6sArJ_mlXfI4cahV2nE,2021
> can/interfaces/nican.py,sha256=y_RPLVSdYgpSIeGQNKrlF6qsntsPh5rDE4t5CDtmiU0,10909
> can/interfaces/pcan/__init__.py,sha256=Vg9ZLRxX6muTOCHqqAdsANC4Zexjg2PRBUs9dA3VgMo,71
> can/interfaces/pcan/__pycache__/__init__.cpython-37.pyc,,
> can/interfaces/pcan/__pycache__/basic.cpython-37.pyc,,
> can/interfaces/pcan/__pycache__/pcan.cpython-37.pyc,,
> can/interfaces/pcan/basic.py,sha256=Gy0wB5z6eRQeojBoj-UAU7QSYggkxADTyYjA2uTYfcU,35390
> can/interfaces/pcan/pcan.py,sha256=5SqkTWHW6WuvSzVqa54gjHAxpGGHTa5A5iPYmVgVXo4,15528
> can/interfaces/serial/__init__.py,sha256=Xje2qNOlrcpJOcLPz3UdU4x5Y7ygRW9Mj63bpmSGoHQ,88
> can/interfaces/serial/__pycache__/__init__.cpython-37.pyc,,
> can/interfaces/serial/__pycache__/serial_can.cpython-37.pyc,,
> can/interfaces/serial/serial_can.py,sha256=WKhAUl035QpQpKC-0R5SjGmj1sMknXxjKBrHJlqcnBk,4913
[... 61 lines cut ...]
> can/notifier.py,sha256=03oT1BaOvYYqqt7eL85OVSapMLl942NZ1bgIA_83fcM,5007
> can/player.py,sha256=8YeosdTPPoA4ka6t8tpEuc2b6rtQdZNQT1_yLTgWOoU,3360
> can/thread_safe_bus.py,sha256=Xp7efg9o8P8MosfDpZwUhOrjtDzOPdtbzklhNAaKT74,3411
> can/util.py,sha256=PpBTlqaO4ywjdNzN1oW_3KZW3CBtT3_NKOl0CMLvx5E,7350
> can/viewer.py,sha256=LLuZq9wJXdecp5L7aM2bOSXr2P9jfG4n2rlGUUdxxZM,22590
> python_can-3.3.2.dist-info/INSTALLER,sha256=zuuue4knoyJ-UwPPXg8fezS7VCrXJQrAP7zeNuwvFQg,4
> python_can-3.3.2.dist-info/LICENSE.txt,sha256=2n6rt7r999OuXp8iOqW9we7ORaxWncIbOwN1ILRGR2g,7651
> python_can-3.3.2.dist-info/METADATA,sha256=nPcEBt8m0ujlcqP2ZlswjvY6R7kWshfOPSa5e7B6TBI,6523
> python_can-3.3.2.dist-info/RECORD,,
> python_can-3.3.2.dist-info/WHEEL,sha256=JZXtYepZFsf4IoivNpnSgKIc4qTHan08DRd56koo_DM,116
> python_can-3.3.2.dist-info/top_level.txt,sha256=tPwJHaIFJDNlJCfW47fReDKSCcRMBXS_rYcARliFueA,9
> test/data/__init__.py,sha256=KwGiZWhwug0K5ngnoP3y1Xx0RPSHLojsRIGCd5uMMB0,38
> test/data/__pycache__/__init__.cpython-37.pyc,,
> test/data/__pycache__/example_data.cpython-37.pyc,,
> test/data/example_data.py,sha256=kJCVCGUVm3ee2vcIe_4DxLI1MN2MSizrRWIHrQ7JZDQ,5472

As you can see it will create a "test" folder in site-packages with your tests.

This results in the following:

since some projects use the deprecated test_suite syntax they will overwrite each other in the site-packages\test
conflicts when having this installed in site-packages and running your own projects tests, which are very often in test folder. it will now run your tests instead of own projects tests
additional unneeded disk space usage and takes longer to install

All in all it's recommended to exclude tests when uploading to pypi.

christiansandberg, 2019-12-06:
It would seem that the test package is correctly excluded but test.data is not. We should add test.* to the exclude list as well.

chrisoro, 2019-12-06:
I already did that, will link PR

karlding, 2020-05-22:
Closing since the PR associated with this is released as part of [v3.3.3](https://github.com/hardbyte/python-can/tree/3.3.3).

closed 2020-05-22 by karlding