hardbyte/python-can #743: ThreadBasedCyclicSendTask stops when any error happens on CAN bus

tysonite, 2019-12-12:
I would like to have a discussion over implementation of `ThreadBasedCyclicSendTask`. The issue there is that if any kind of error happens on the CAN bus, it terminate itself:

https://github.com/hardbyte/python-can/blob/b0d74607de0e411ea3b19347cee20cf8ccba0dcd/can/broadcastmanager.py#L251-L253

I see few issues there:
1. The user (application code actually) is not notified if error happens (`bus.send()` fires exception) during sending periodic messages in `ThreadBasedCyclicSendTask`, thus application can't handle it properly, for example, restart periodic task after some time. Do I miss something and is this possible?

2. It contradicts with a behavior of, for example, SocketCAN implementation of `send_periodic` as it moves responsibility of sending periodic messages to the kernel level, and it is expected that error handling will be done inside the kernel. As tests show, if error on CAN bus happens during usage of native periodic implementation, the periodic tasks are still there and it covers after some time.

Any thoughts on this?

christiansandberg, 2019-12-16:
It could certainly be discussed. I’m not sure how to handle that situation though. Letting the developer decide is probably better than trying to force some recovery mechanism that won’t suit everyone.

tysonite, 2019-12-17:
I would think about having a callback function passed as an argument to `bus.send_periodic`, it may accept an exception raised internally by `ThreadBasedCyclicSendTask` so application can be implement any custom logic based on that.

tysonite, 2020-05-25:
I could close this issue as fix is merged already. Or is there anything else in your minds?

closed 2020-05-27 by tysonite