hardbyte/python-can #746: Tests fail on develop

marcel-kanter, 2019-12-29:
Using commit dcd694efb56d503ed9387edc5cfba9c523d559bf the tests fail.
Windows 10 Pro Version 1909
Python 3.7.6, pytest-5.3.2, py-1.8.1, pluggy-0.13.1
pytest plugins: timeout-1.3.3, cov-2.8.1, hypothesis-4.57.0

output of python setup.py test:

test/test_message_class.py::TestMessageClass::test_methods PASSED        [ 62%]
test/test_message_filtering.py::TestMessageFiltering::test_match_all PASSED [ 62%]
test/test_message_filtering.py::TestMessageFiltering::test_match_example_message PASSED [ 63%]
test/test_message_filtering.py::TestMessageFiltering::test_match_filters_is_empty PASSED [ 63%]
test/test_message_sync.py::TestMessageSync::test_general PASSED          [ 64%]
+++++++++++++++++++++++++++++++++++ Timeout ++++++++++++++++++++++++++++++++++++

After commenting out the pytest timeout detorator for the test, it results in:

test/simplecyclic_test.py::SimpleCyclicSendTaskTest::test_cycle_time FAILED [ 42%]
test/test_scripts.py::TestLoggerScript::test_do_commands_exist FAILED    [ 72%]
test/test_scripts.py::TestPlayerScript::test_do_commands_exist FAILED    [ 73%]
================= 3 failed, 194 passed, 29 skipped in 36.53s ==================

marcel-kanter, 2019-12-29:
N.B. During writing some CANopen python code and switching between OSes I noticed that Windows 10 is a bit slower than Windows 7...

Regarding the timeout decorator:
The timeout for the test should NOT be used as test criterea. If there is a timing requirement for a test, then it should be tested within the test. So the timeout decoration should be removed, the time increased or the timing requirement (I don't see one here) be moved into the test code.

marcel-kanter, 2020-01-04:
test/simplecyclic_test.py::SimpleCyclicSendTaskTest::test_cycle_time
fails because  there are not messages passing through. I got only 51, sometimes 66 of 100.
How much is this in non ci linux?

christiansandberg, 2020-01-05:
Could be worth investigating if we can use [timeBeginPeriod](https://docs.microsoft.com/sv-se/windows/win32/api/timeapi/nf-timeapi-timebeginperiod) to increase timer resolution on Windows.

marcel-kanter, 2020-01-05:
I think it's not the timer resolution, I'll try to investigate in what happens at every time step.
Because the thread switching time on my small Odroid C2 is below 1ms (more or less in 0.2ms range), but something makes the timing bad on the CI maschines.
Additionally i found that under Ubuntu 18.04 LTS the test fail too:
[odroidc2.txt](https://github.com/hardbyte/python-can/files/4023494/odroidc2.txt)

christiansandberg, 2020-01-05:
The CI machines are heavily loaded by lots of tasks so it is useless running timing critical tests on those.

marcel-kanter, 2020-01-05:
ok

marcel-kanter, 2020-01-09:
Ok. Got it fixed. There is a problem with the Python 3.7.4.1 install on Windows which comes with a module typing.py and this is the source of the bug.
To fix this typing issue do:
python -m pip uninstall typing

closed 2020-02-02 by marcel-kanter