hardbyte/python-can #751: ASC log file reader and writer does not write DLC for RTR frames.

marcel-kanter, 2020-01-01:
The ASCWriter does not write the DLC for RTR frames.
RTR frames require an DLC. Vectors tools write the DLC as single (and only) data byte, but ASCWriter does not.
Using CANalyzer 12 SP4, the log for a RTR TX mesage will be generated:

date Wed Jan 1 10:27:12.172 pm 2020
base hex  timestamps absolute
internal events logged
// version 12.0.0
// Measurement UUID: cc670aa7-6a8a-4b39-bae0-260bad033bbf
Begin TriggerBlock Wed Jan 1 10:27:12.172 pm 2020
   0.000000 Start of measurement
   1.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
   1.210098 1  1               Tx   r 8  Length = 92000 BitCount = 49 ID = 1
   2.000000 1  Statistic: D 0 R 1 XD 0 XR 0 E 0 O 0 B 0.00%
   3.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
   4.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
End TriggerBlock

Manipulating the file into Rx frame, removing the DLC and replaying the log through CANalyzer again shows that the DLC is generated again and the missing DLC is assumed to be 0:

date Wed Jan 1 10:15:34.519 pm 2020
base hex  timestamps absolute
internal events logged
// version 12.0.0
// Measurement UUID: e287c73d-66b3-4eb0-9173-3e847541542a
Begin TriggerBlock Wed Jan 1 10:15:34.519 pm 2020
   0.000000 Start of measurement
   1.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
   2.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
   3.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
   4.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
   5.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
   6.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
   7.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
   8.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
   9.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
   9.790098 1  1               Rx   r 0  Length = 92000 BitCount = 49 ID = 1
  10.000000 1  Statistic: D 0 R 1 XD 0 XR 0 E 0 O 0 B 0.00%
  11.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
  12.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
  13.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
  14.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
  15.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
  16.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
  17.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
  18.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
  19.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
  20.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
  21.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
  22.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
  23.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
  24.000000 1  Statistic: D 0 R 0 XD 0 XR 0 E 0 O 0 B 0.00%
End TriggerBlock

marcel-kanter, 2020-01-01:
Additionally the DLC is in hex format without leading 0x, not in decimal

marcel-kanter, 2020-01-10:
I found out that this is a bug in vectors file format. Canalyzer v 7 has this bug. v12 not. Somewhere in between they changed this. I'll make the reader compatoble by defaulting a missimg dlc to 0 as this is the behavior in their sw and the real value is not available.

felixdivo, 2021-04-23:
Is this still an issue that needs fixing? Is it reproducible in version `4.0.0-dev.2`?

Zennn11zen11, 2025-01-29:
http://www.blockexplorer.com/address/THEADDRESSGOESHERE

Zennn11zen11, 2025-01-29:
0x0c62ebe56010dfaf7885b78dd91cba243ddd6866