hardbyte/python-can #752: SLCAN, Robotell, socketcan, serial, seeedstudio use fileno property-like function for i/o

marcel-kanter, 2020-01-04:
In the interfaces SLCAN, Robotell, socketcan, serial and seeedstudio there is a property-like function called fileno. This is not marked as property and is used in notifier test for i/o.
As written in the BusABC description there should be no "custom" property.
Using this method leaves a coverage gap on Windows AND on Linux.

The notifier will not use the asyncio loop in add_bus but tries to clear the loop in stop and do not join the reader.

The special handling for these interfaces should be removed and only the recv method used (as in all other interfaces) and the notifier reworked.

christiansandberg, 2020-01-04:
I didn’t quite understand what the problem was. Is it that the method is not covered by unit tests?

The fileno() method is a convention for classes that have a file descriptor. It allows them to be monitored for changes in an event loop directly instead of having to create an unnecessary thread.

It would probably be cleaner to implement this method in BusABC as well which would raise OSError or return -1.

marcel-kanter, 2020-01-04:
The problem is that this is not covered in the tests correctly.

christiansandberg, 2020-01-04:
In general we don’t need to test trivial getter and setter methods.

marcel-kanter, 2020-01-04:
It's not tested for the notifier and not testing trivial getters leaves the door open for errors..

christiansandberg, 2020-01-04:
You’re of course free to write tests for those if you really want to. But staring too blindly at 100% code coverage is not necessarily the best way to feel confident in the code base.

https://martinfowler.com/bliki/TestCoverage.html

marcel-kanter, 2020-01-04:
Yes, of course Martin is absolutely right. And no, i found this by code review, and as a consequence of found issues there should be additional tests that cover that issue. They should fail before the fix and pass after the fix.
Testing like a stupid with stupid data is stupid. ;)