hardbyte/python-can #757: Increased filter ability

vitali-mueller, 2020-01-20:
Hello guys,

I'm working with a CAN-shield to read out HEX-Codes within an electric vehicle (socketcan).
I sent all the HEX-Codes I needed, and I get a correct response after manually transform the HEX-code to a decimal number.
For example I expect following received message:
`can_id: 0x607`
`data: [0xF1 , 0x06, 0x62, 0xDD, 0xC4 0x22, 0xC9 , 0x01]`

So first approach with the bus.set_filters was working.
`bus.set_filters([{"can_id": 0x607, "can_mask":0xF1, "extended": False}])`

Only the messages from the bus did arrive, but also some other unrelevant messages, which disturb my following system.
I tried to use an array for the can_mask but it just delivers me an error.
Is there a possibility to extend the filter, so I can filter more bytes? Like `[0xF1 , 0x06, 0x62, 0xDD]`?
And for information, if I change the can_mask to 0xF4 or another number in the end, the filter still gives me back 0xF1.

For my project, I want to have only the HEX-Code as response without any other things like timestamp or DLC. Is there a possibility to delete these columns or only grab the HEX-Code?
I saw following code in socketcan.py row 195 (new code at row 172):
`data = bytes(msg.data).ljust(max_len, b'\x00)`
Is this somehow possible to use this code to get the answer from the bus only, instead of `bus.recv()`?

I'm a bloody python newbie btw...

Thanks and best regards,
Vitali

christiansandberg, 2020-01-21:
I think you’ve misunderstood the can_mask. It defines which bits in the can_id that should be matched. If in your case you only want 0x607 then the mask should be 0x7ff. There is no way to filter on data, that you’ll have to do yourself when the message has been received.

vitali-mueller, 2020-01-23:
Thanks for your reply.
Seems like I still miss some information.
How do you get 0x7ff for the can_mask?

christiansandberg, 2020-01-23:
You can read how acceptance filters work here for example:
https://www.kvaser.com/canlib-webhelp/page_user_guide_send_recv.html#section_user_guide_send_recv_filters

Basically a message would match if the CAN ID & 0x7ff == 0x607. 0x7ff means all 11 bits of a standard CAN ID must match.

karlding, 2020-04-23:
Closing this out as there doesn't seem to be anything actionable here. Please reopen if necessary.

closed 2020-04-23 by karlding