hardbyte/python-can #758: help with a config

maltsevvv, 2020-01-21:
config with Python-can1.4.3 works.
```
#!python
#
from __future__ import print_function
import time
import os
import sys
import threading
import can
can.rc['interface'] = 'socketcan_ctypes'
from threading import Thread, Timer
from can.interfaces.interface import Bus
can_interface = 'can0'

global var
var=1

def dumpcan():
    global var
    for message in Bus(can_interface):
        if var==1:
            msg = unicode(message).encode('utf +8')
            canid = msg[26:29]
            msg = msg[45:69]
            if canid == ("464"):
                if msg == ("e0 01 00"):
                    bus = can.interface.Bus()
                    msg = can.Message(arbitration_id=0x264, data=[0xa0, 0x01, 0x00], extended_id=False)
                    bus.send(msg)
                if msg == ("a1 01"):
                    bus = can.interface.Bus()
                    msg = can.Message(arbitration_id=0x264, data=[0x10, 0x15, 0x01, 0x00, 0x02, 0x00, 0x00], extended_id=False)
                    bus.send(msg)
dumpcan()
```  

 I ask for your help, with a change for python-can 3.3.2   

` from can.interfaces.interface import Bus ImportError: No module named interface`

and how to add a parameter here. if there are no messages from 464, then send 01010101 only once?

Sorry to ask here. And thanks in advance

bessman, 2020-01-27:
If I understand you correctly, you want to do the following:
```
Monitor the bus for messages with ID 0x464
IF message data is [0xe0, 0x01, 0x00], then
    Send a reply message with ID 0x264 containing data [0xa0, 0x01, 0x00]
ELSE IF message data is [0xa1, 0x01], then
    Send reply message with ID 0x264 containing data [0x10, 0x15, 0x01, 0x00, 0x02, 0x00, 0x00]
IF there are no messages with ID 0x464 on the bus, then
    Send a message with ID 0x264 containing [0x00, 0x01, 0x00, 0x01, 0x00, 0x01, 0x00, 0x01] once
```
Is that right? If so, this will work:
```
import can

def dumpcan(timeout=5):
    # Initialize a CAN bus with a filter to ignore all messages with IDs other
    # than 0x464.
    filt = [{"can_id": 0x464, "can_mask": 0x7ff, "extended": False}]
    bus = can.Bus(bustype="socketcan", channel="can0", can_filters=filt)

    try:
        while True:
            # Monitor the bus for messages. If no messages are receieved within
            # <timout> seconds, message will be None.
            message = bus.recv(timeout=timeout)

            if message is None:
                # If there are no messages with ID 0x464 on the bus, send once
                # and then exit.
                msg = can.Message(
                    arbitration_id=0x264,
                    data=[0x00, 0x01, 0x00, 0x01, 0x00, 0x01, 0x00, 0x01],
                    extended_id=False
                )
                bus.send(msg)
                break
            elif list(message.data) == [0xe0, 0x01, 0x00]:
                msg = can.Message(
                    arbitration_id=0x264,
                    data=[0xa0, 0x01, 0x00],
                    extended_id=False
                )
                bus.send(msg)
            elif list(msg.data) == [0xa1, 0x01]:
                msg = can.Message(
                    arbitration_id=0x264,
                    data=[0x10, 0x15, 0x01, 0x00, 0x02, 0x00, 0x00],
                    extended_id=False
                )
                bus.send(msg)
            else:
                # If a message with ID 0x464 is received containing any data
                # other than [0xe0, 0x01, 0x00] or [0xa1, 0x01], then do
                # nothing.
                pass
    finally:
        # Deinitialize the bus
        bus.shutdown()

dumpcan()
```

maltsevvv, 2020-01-27:
Thanks. and how else to adapt my config, under python-can 3.3, this one works with 1.8.4. I read manuals, but I'm not a programmer

bessman, 2020-01-28:
Well, if your script works well with python-can 1.4.3, then perhaps stick with that version? Provided you don't need any new features introduced since then, of course. If you do, then take a look at the [documentation](https://python-can.readthedocs.io/en/stable/). If you find it unclear, or if it's missing something, you can open an issue asking for clarification.

maltsevvv, 2020-01-28:
OK. thanks

closed 2020-01-28 by maltsevvv