hardbyte/python-can #759: OSError: [WinError 126] The specified module could not be found - PCAN application

natek512, 2020-01-27:
Hello all! Noob here running into an issue even creating the Bus object.
I've create a virtual environment with the python-can module installed. I tried creating a Bus object via the following

```
import can

bus = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=250000)
```

and this is the error I receive:

```
runfile('C:/Documents/Python Scripts/can_api_tutorial.py', wdir='C:Documents/Python Scripts')
Traceback (most recent call last):

  File "C:\Documents\Python Scripts\can_api_tutorial.py", line 17, in <module>
    bus = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=250000)

  File "C:\.conda\envs\CANenv\lib\site-packages\can\interface.py", line 127, in __new__
    return cls(channel, *args, **config)

  File "C:\.conda\envs\CANenv\lib\site-packages\can\interfaces\pcan\pcan.py", line 97, in __init__
    self.m_objPCANBasic = PCANBasic()

  File "C:\.conda\envs\CANenv\lib\site-packages\can\interfaces\pcan\basic.py", line 359, in __init__
    self.__m_dllBasic = windll.LoadLibrary("PCANBasic")

  File "C:\.conda\envs\CANenv\lib\ctypes\__init__.py", line 426, in LoadLibrary
    return self._dlltype(name)

  File "C:\.conda\envs\CANenv\lib\ctypes\__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)

OSError: [WinError 126] The specified module could not be found
```
Pretty new to this so sorry if this is a silly question Thanks!

bmeisels, 2020-01-27:
It seems python-can can't find PCANBasic.dll.
Have you installed the pcan driver package. It can be downloaded from the PEAK-System website.

natek512, 2020-01-28:
Thanks a lot! I didn't have the PCAN-Basic option checked when I went through my first installation. I'm able to compile now.

closed 2020-01-28 by natek512