hardbyte/python-can #767: I can't get the motor moving

Gius-8, 2020-02-12:
Hello,
thanks to this library I was able to read and write the parameters of the motor in canopen, but I am new in canopen and I have not yet understood the procedure to follow to get the motor moving. Can someone help me?
Thanks so much

bessman, 2020-02-14:
> Can someone help me?

Not without knowing what type of motor it is. Even then, this is not the correct forum for this question unless the reason you cannot get the motor moving is a bug in python-can.

You will probably want to use a CANopen library to communicate with your device, rather than using raw CAN directly. Try this: https://github.com/christiansandberg/canopen

karlding, 2020-04-13:
Closing this out as there doesn't seem to be anything anything actionable here. Please reopen if necessary.

closed 2020-04-13 by karlding