hardbyte/python-can #771: CSVReader crashes when reading an empty file

Krensi, 2020-02-20:
The next statement in the __iter__ function of the class CSVReader is not inside the for loop. There for a Stopiteration error is not catched.

Rewriting the code like this fixes this:
`    
def __iter__(self):
        for (index, line) in enumerate(self.file):
            if index > 0: # Skip header line
                timestamp, arbitration_id, extended, remote, error, dlc, data = line.split(',')

                yield Message(
                    timestamp=float(timestamp),
                    is_remote_frame=(remote == '1'),
                    is_extended_id=(extended == '1'),
                    is_error_frame=(error == '1'),
                    arbitration_id=int(arbitration_id, base=16),
                    dlc=int(dlc),
                    data=b64decode(data),
                )
`

bessman, 2020-02-21:
This is [fixed](https://github.com/hardbyte/python-can/blob/b87329983d03b91b16920d47a8b4f90e650ca433/can/io/csv.py#L91) in develop:
```       
        # skip the header line
        try:
            next(self.file)
        except StopIteration:
            # don't crash on a file with only a header
            return
```

karlding, 2020-05-22:
Closing since the fix is released as part of [v3.3.3](https://github.com/hardbyte/python-can/tree/3.3.3).

closed 2020-05-22 by karlding