hardbyte/python-can #775: Stop Notifier from raising an exception if it has been handled by on_error

Aajn, 2020-02-21:
Currently, when an exception is raised by a bus (e.g. bus-off) the Notifier allows listeners to handle the error with the on_error method. In the Listener documentation I can see that

> This method is called to handle any exception in the receive thread

But this does not stop the Notifier from raising an exception anyway.

Would it make sense to stop Notifier from raising an Exception if on_error is called or returns True, to stop this behavior?

christiansandberg, 2020-02-21:
I think your suggestion for returning True to prevent Notifier from stopping sounds good. In my experience a receive exception is usually fatal and won’t heal itself without creating a new Bus object, but more choice is better.

Aajn, 2020-02-22:
My main problem with exceptions being raised in the thread is that it causes the threat to stop ungracefully and print a stack trace (it might also affect the exit code, but I cannot verify this right now).

Maybe a better solution is to treat on_error as an exception handler. If implemented, it could be called stopping the exception from being raised again. On the other hand, you don’t have control over the kind of Exception you want to handle.

Which approach do you think is better in the context of the library?

I would have time to prepare a PR some time next week.

closed 2020-05-30 by Aajn