hardbyte/python-can #776: Supporting Tx/Rx message direction for each interface

sou1hacker, 2020-02-21:
Pull request #773 adds is_rx attribute to can.Message. Currently this attribute is set to True by default. Functionality needs to be added to each interface type to set this attribute appropriately when building CAN messages:

- [x] ics_neovi
- [ ] ixxat
- [ ] kvaser
- [ ] pcan
- [ ] seedstudio
- [ ] serial
- [ ] socketcan
- [ ] systec
- [ ] usb2can
- [ ] vector
- [ ] nican
- [ ] iscan
- [ ] virtual
- [ ] slcan
- [ ] robotell
- [ ] canalystii

bessman, 2020-02-24:
How should `receive_own_messages` behave with this change? Should messages transmitted by other applications have `is_rx=False`?

I'm looking at Kvaser's canlib, and I can see no obvious way to accomplish that.

tamenol, 2020-02-27:
I was looking to implement this for the PCAN bus. However I can't seem to find a variable in the PCANbasic library I could use to check if it's RX or TX.  Someone has an idea?

`TPCANMessageType` only has these values:
```python
# PCAN message types
PCAN_MESSAGE_STANDARD = TPCANMessageType(
    0x00
)  # The PCAN message is a CAN Standard Frame (11-bit identifier)
PCAN_MESSAGE_RTR = TPCANMessageType(
    0x01
)  # The PCAN message is a CAN Remote-Transfer-Request Frame
PCAN_MESSAGE_EXTENDED = TPCANMessageType(
    0x02
)  # The PCAN message is a CAN Extended Frame (29-bit identifier)
PCAN_MESSAGE_FD = TPCANMessageType(
    0x04
)  # The PCAN message represents a FD frame in terms of CiA Specs
PCAN_MESSAGE_BRS = TPCANMessageType(
    0x08
)  # The PCAN message represents a FD bit rate switch (CAN data at a higher bit rate)
PCAN_MESSAGE_ESI = TPCANMessageType(
    0x10
)  # The PCAN message represents a FD error state indicator(CAN FD transmitter was error active)
PCAN_MESSAGE_ERRFRAME = TPCANMessageType(
    0x40
)  # The PCAN message represents an error frame
PCAN_MESSAGE_STATUS = TPCANMessageType(
    0x80
)  # The PCAN message represents a PCAN status message
```

chrisoro, 2020-06-22:
@hardbyte @pierreluctg 
any indication when you release a version based on the develop branch? this feature is really interesting for us (vector devices for now)