hardbyte/python-can #777: pcan timestamp issue

psedlos, 2020-02-24:
Hi there, I have strange issue with time for received messages. Please check code below:
import can
import time

bustype = 'pcan'
channel = 'PCAN_USBBUS1'
bitrate = 250000

bus = can.interface.Bus(bustype = bustype, channel = channel, bitrate = bitrate)
msg = bus.recv(1)
print(msg)
diff = msg.timestamp - time.time()
print(diff)

The result should be around 0 but it is around 3600 which drive me directly to timezone, which is at my PC: time.timezone = -3600.
However I can not find any information about configuring timezones for CAN, and I believe it should not be possible to do it.

Therefore here is my question. Is python-can wrong here or I miss something?

bessman, 2020-02-24:
This looks like a bug in pcan:
https://github.com/hardbyte/python-can/blob/cb138c09567b9502d1040b93ad2af9c16b0cafd7/can/interfaces/pcan/pcan.py#L19-L21
`uptime.boottime()` takes the timezone into account but `datetime.utcfromtimestamp()` obviously does not. This creates a spurious timedelta equal to the timezone. `datetime.fromtimestamp()` should be used instead.

tiger86023086, 2021-04-27:
> This looks like a bug in pcan:
> https://github.com/hardbyte/python-can/blob/cb138c09567b9502d1040b93ad2af9c16b0cafd7/can/interfaces/pcan/pcan.py#L19-L21
>
>
> `uptime.boottime()` takes the timezone into [...]

I use the script'boottimeEpoch = ( 
     uptime.boottime() - datetime.datetime.fromtimestamp(0) 
 ).total_seconds() '

but there is error.The error is:
Traceback (most recent call last):
  File "D:\Python37\lib\site-packages\can\interface.py", line 36, in _get_class_for_interface
    module = importlib.import_module(module_name)
  File "D:\Python37\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "D:\Python37\lib\site-packages\can\interfaces\pcan\__init__.py", line 4, in <module>
    from can.interfaces.pcan.pcan import PcanBus
  File "D:\Python37\lib\site-packages\can\interfaces\pcan\pcan.py", line 21, in <module>
    uptime.boottime() - datetime.datetime.fromtimestamp(0)
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "E:\Project\CCM_Test\456.py", line 2, in <module>
    bus = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=500000)
  File "D:\Python37\lib\site-packages\can\interface.py", line 98, in __new__
    cls = _get_class_for_interface(kwargs["interface"])
  File "D:\Python37\lib\site-packages\can\interface.py", line 40, in _get_class_for_interface
    module_name, interface, e
ImportError: Cannot import module can.interfaces.pcan for CAN interface 'pcan': [Errno 22] Invalid argument

Is there is error or Am i use it in wrong?

bessman, 2021-04-27:
`OSError: [Errno 22] Invalid argument` is generally related to file access, typically trying to open incorrectly formatted paths. Please post a small piece of sample code which reproduces the issue. For example, does the following generate the same error?
```python
import can
bus = can.Bus(interface="pcan")
```

tiger86023086, 2021-04-29:
> `OSError: [Errno 22] Invalid argument` is generally related to file access, typically trying to open incorrectly formatted paths. Please post a small piece of sample code which reproduces the issue. [...]

I just use a small piece of code.

import can
bus = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=500000)

The code will genenrate error 22

bessman, 2021-04-29:
And the error is still reported as occurring at `uptime.boottime() - datetime.datetime.fromtimestamp(0)`? If so, does simply running `uptime.boottime()` or `datetime.datetime.fromtimestamp(0)` individually cause an error?

tiger86023086, 2021-04-29:
Now my python version is python37(64).

>>> import datetime
>>> datetime.datetime.fromtimestamp(0)
Traceback (most recent call last):
  File "<pyshell#5>", line 1, in <module>
    datetime.datetime.fromtimestamp(0)
OSError: [Errno 22] Invalid argument
>>>

I find two library 'datetime' in my python37.One path is 'D:\Python37\Lib\datetime.py',Another path is 'D:\Python37\Lib\site-packages\future\backports\datetime.py'.

I find the function as follow:
1、
@classmethod
    def fromtimestamp(cls, t, tz=None):
        """Construct a datetime from a POSIX timestamp (like time.time()).

        A timezone info object may be passed in as well.
        """
        _check_tzinfo_arg(tz)

        return cls._fromtimestamp(t, tz is not None, tz)

2、
@classmethod
    def fromtimestamp(cls, t, tz=None):
        """Construct a datetime from a POSIX timestamp (like time.time()).

        A timezone info object may be passed in as well.
        """

        _check_tzinfo_arg(tz)

        converter = _time.localtime if tz is None else _time.gmtime

        t, frac = divmod(t, 1.0)
        us = int(frac * 1e6)

        # If timestamp is less than one microsecond smaller than a
        # full second, us can be rounded up to 1000000.  In this case,
        # roll over to seconds, otherwise, ValueError is raised
        # by the constructor.
        if us == 1000000:
            t += 1
            us = 0
        y, m, d, hh, mm, ss, weekday, jday, dst = converter(t)
        ss = min(ss, 59)    # clamp out leap seconds if the platform has them
        result = cls(y, m, d, hh, mm, ss, us, tz)
        if tz is not None:
            result = tz.fromutc(result)
        return result

I do not knonw to which one of name 'datetime' lib

bessman, 2021-04-30:
You are using the first one. The second would need to be imported as `from future.backports import datetime`.

Your issue is not a bug in python-can, but rather in Python itself: https://bugs.python.org/issue29097

It should be fixed already, according to the tracker. Are you using the latest version of Python3.7? Regardless, since the issue is unrelated to python-can I suggest we close this again.

felixdivo, 2021-04-30:
@bessman Thanks for assisting! Indeed, it seems like there's nothing to be fixed by us here.

PR #778 (2020-06-01): PCAN: Fix timestamp timezone offset
Fix #777
closed 2021-04-30 by felixdivo