hardbyte/python-can #782: Question: Having a node that listens for messages (Listener, Notifier)

aHudspith, 2020-02-28:
Hi,

I'm pretty new to this. I'm wanting to use this module to build a virtual CAN network representing part fo the CAN network of a car. It's going to be very basic, I want to have two nodes that can send messages, and one that just receives. Effectively it's going to be a node representing the break pedal, a node representing an attacker and a node representing the engine (I'm aware this is not how it works in an actual car). I'm struggling to grasp how to do this with the module. I feel like I want the "engine" to be a listener, but I'm struggling to see how I actually do this. I've looked at the  examples but they haven't seemed to help.

Any help would be greatly appreciated :)

bessman, 2020-02-28:
You could use the virtual bus type. You want all your nodes to be on the same network, correct? It could look something like this:
```
import can
import time

brakes_node = can.Bus(channel="my_bus", bustype="virtual")
engine_node = can.Bus(channel="my_bus", bustype="virtual")
attacker_node = can.Bus(channel="my_bus", bustype="virtual")

engine_listener = can.Printer()
engine_notifier = can.Notifier(bus=engine_node, listeners=[engine_listener])

slow_down = can.Message(arbitration_id=0x1, data=[0])
speed_up = can.Message(arbitration_id=0x1, data=[255])

brakes_node.send_periodic(msgs=slow_down, period=.5)
attacker_node.send_periodic(msgs=speed_up, period=.1)

time.sleep(5)

# Cleanup
engine_notifier.stop()
engine_node.shutdown()
```
Here, I create three nodes and connect them all to the same virtual bus. This means than any message sent by any of the three will be received by the other two.

I then create a Listener (a Printer in this case, but it could be any type of Listener) and a Notifier for the engine node. The engine_notifier will pass any message received by the engine_node (i.e. any message sent by either brakes_node or attacker_node) to the engine_listener, which will print it to the terminal.

I then create two messages with the same arbitration ID; one says to slow down, the other to speed up. The brakes_node sends the slow_down message twice per second, while the attacker sends the speed_up message ten times per second. You should see these messages appear in your terminal.

Hope that helps!

felixdivo, 2020-04-29:
Seems like the kind explanation of @bessman helped. I will close this for now.

closed 2020-04-29 by felixdivo