hardbyte/python-can #786: BLF padding/alignment incorrect for CAN_FD_MESSAGE_64

MartinStellmacher, 2020-03-04:
In blf.py all messages are padded by obj_size % 4. This assumption doesn't hold for all types of messages. I experienced a parsing error while reading a file which contained CAN_FD_MESSAGE_64 items. They are not padded at all, as you can see in Toby Lorenz C++ implementation.

Issue #459 is closed, but the problem still persists. Solution might be as simple as:

                    if header[4] == CAN_FD_MESSAGE_64 or header[4] == ETHERNET_FRAME_EX: 
                        next_pos = pos + obj_size #no padding
                    else:
                        next_pos = pos + obj_size + (obj_size % 4)  #weird padding

In Toby Lorenz implementation the padding is handled separately for each frame type...

christiansandberg, 2020-03-05:
This fix (for FD frames only) is not released, you need to use develop.

JuicyPussy, 2020-03-06:
faced the same problem, how could I contribute to fix it?

christiansandberg, 2020-03-06:
A log file would be helpful. You can e-mail it to me if you don’t want to share it publicly: christiansandberg at me dot com.

MartinStellmacher, 2020-03-06:
Hi Christian, thank you, but I think this fix isn't enough. In the BLF files I have at hand I found three different kinds of padding: 
1. none, like for the CAN_FD, 
2. crazy padding with `length%4` (mostly used) and 
3. correct padding with `(4-length%4)%4` (I think these were Ethernet-Frames). Even if you don't wan't to read them, your parser will fail, if they are contained within the file.

My final solution is to assume no padding and correct it by searching the LOBJ-mark with offsets from 0-3. Doing it this way I can separate the container format completely from the payload formats. I couldn't measure any increase in parsing times.

christiansandberg, 2020-03-06:
Yeah that’s probably the best solution. Just need to make sure it handles padding spanning over two containers.

MartinStellmacher, 2020-03-06:
Yes, I would do the check before reading the header, after fixing the start again for the header and than again to ensure that the payload is in the same container.

christiansandberg, 2020-03-07:
If you could make a pull request with your solution that would be awesome.

PR #806 (2020-04-19): Fix BLFParseError
Seek for start of object instead of calculating it

Fixes #803
Fixes #786
commit 77802421 (2020-04-19): Seek for start of object instead of calculating it (#806)

Fixes #803
Fixes #786
closed 2020-04-19 by hardbyte