hardbyte/python-can #787: default can message pass extended_id=None, will make message to be extended

zhupumpkin, 2020-03-06:
in file can\message.py
line 83 function __init__ for Message 
pass is_extended_id=None also extended_id=None which are all defaul. 
And from meaning, it means no extended.
However line 113 in the same file 
            self.is_extended_id = True if extended_id is None else extended_id
this logic makes extended true as extended_id is None which is logical error.

expect should be 
self.is_extended_id = True if extended_id is not 
[image]
None else extended_id

christiansandberg, 2020-03-06:
Older versions had extended_id which was True default. Then it was renamed to is_extended_id but to keep backwards compatibility the old was kept and we allowed both during a transition period. That’s why it is a little complicated.

felixdivo, 2020-04-29:
The current version on master already has this simplified and dropped backwards compatibility, see [here](https://github.com/hardbyte/python-can/blob/develop/can/message.py#L81). So I think there is no need for action here.

closed 2020-08-06 by hardbyte