hardbyte/python-can #791: Seeedstudio USB-CAN interface not present

davidklett, 2020-03-12:
Hello, I get the following message when I run the following command on my command line in Windows:
pip3 install python-can[seeedstudio]

WARNING: python-can 3.2.0 does not provide the extra 'seeedstudio'

Because of this, I'm not able to run the following code:
bus = can.interface.Bus(bustype='seeedstudio', channel='/dev/ttyUSB0', bitrate=500000)

christiansandberg, 2020-03-13:
This interface is not present in any released version. You need to use develop branch for now.

@hardbyte @felixdivo maybe it’s time to make a release soon?

felixdivo, 2020-03-13:
Yeah, why not. Maybe we can clean up the [4.0 Release milestone](https://github.com/hardbyte/python-can/milestone/8) and then work on the remaining issues that we want to resolve.

closed 2020-05-20 by davidklett

felixdivo, 2022-01-15:
> Yeah, why not.

Huh, this did not age well ... (#1133)