hardbyte/python-can #793: not an issue but a curiosity!

Soodep, 2020-03-13:
I am trying to sent a cyclic can message from my NVIDIA PX2 through can0. If anyone can help me with that that would be very helpful. Thank you.

karlding, 2020-03-15:
I don't own a PX2, but I'm assuming `can0` is a SocketCAN interface, in which case you can instantiate a bus using the SocketCAN backend as the interface. You can then send periodic messages using the [`send_periodic` API](https://python-can.readthedocs.io/en/master/bus.html#can.BusABC.send_periodic) that operates using the Linux kernel's provided BCM support.

Does that answer your question?

Soodep, 2020-03-15:
Yes it helps a lot. Thank you. I will ask more questions if I get into more issues/problem. Thank you so much for the reply. Appreciate it.

closed 2020-03-15 by karlding