hardbyte/python-can #794: Not Issue , but a help about how to Record the CAN BUS log

CSG2019, 2020-03-15:
Hello,
I am pretty new about this Python-can. 
I just send successfully CAN message  to the bus with this library.  Now I want to know how to get the real bus from vehicle.
I read the official document, but i don't understand it well. It seems that i should use can.Linster to record the CAN bus. 
I want to know how to record the actual vehicle bus(only need channel 1). it's better to replay in CANoe/CANalyzer. Because my hardware is Vector tool.
bus = can.interface.Bus(bustype='vector', app_name='CANoe', channel=[0])
If you can give me some examples or suggestions, I would really appreciate it!!!

bessman, 2020-03-16:
If you all you want to do is log messages to a file (without acting on the message contents), you can use the included [logging script](https://python-can.readthedocs.io/en/master/scripts.html#can-logger):
```
python -m can.logger --file_name="can_log.asc" --channel=0 --interface="vector"
```
The ASC file type is a vector format and works well with their tools.

CSG2019, 2020-03-19:
thank you for your help. now i know how to do it.

closed 2020-03-19 by CSG2019