hardbyte/python-can #797: CAN Interface Module

CSG2019, 2020-03-20:
Dear Friends,

About the CAN Interface module, I have 2 questions as below:

1) I tried to connect the VN1630 of Vector's product, but it failed and the error was : "vector.exceptions.VectorError",  the code is :
import can
bus = can.interface.Bus(bustype='vector', app_name='CANalyzer', channel=0)
I am sure the connection of hardware was sucessful. 

2) Can I ues the ES581 of ETAS product? If yes, which bustype and app_name can I use?
bus = can.interface.Bus(bustype=???, app_name=??? channel=0,1,?)

If you can help me, I would really appreciate it.
thank you!

CSG2019, 2020-03-27:
is there anybody can help me find why the Vector interface connected failed, my hardware is VN1630.
the code is:
`import can
bus_ = can.interface.Bus(bustype='vector', app_name='CANalyzer', channel=0, bitrate=250000)`
the error is :
can.interfaces.vector.exceptions.VectorError: xlGetChannelIndex failed (XL_ERR_HW_NOT_PRESENT)

but i can read and find VN1630(also serial number)in vector hardware config
so it confuses me.

thank you very much.

zariiii9003, 2020-04-11:
Did you assign the first CANalyzer channel to your VN1630 in vector hw config?

CSG2019, 2020-04-12:
> Did you assign the first CANalyzer channel to your VN1630 in vector hw config?

Hi, Thank you for the reply.
I didn't do any operation in Vector Hardware config, it's always default config, but i check the Vector Hardware Config, I can find my VN1630 is present.
when I connect the VN1640 with CANoe License, it is also default config, but it can work well.
I even reinstalled the python-can library.

it confuses me for a long time.
if you can help me find some clues, i am very grateful.

the error is :

> 
can.interfaces.vector.exceptions.VectorError: xlGetChannelIndex failed (XL_ERR_HW_NOT_PRESENT)
> 

thank you again.

zariiii9003, 2020-04-12:
It should look like this (my interface is not plugged in right now):

[image]

CSG2019, 2020-04-12:
> It should look like this (my interface is not plugged in right now):
> 
> [image]

Hello zariiii9003,

thank you very much indeed for your answers. :) 
After i assigned the CANalyzer CAN1 for VN1630 in Vector Hardware Config according your configuration, it really works now.
is there any way to use for the default mode?
below is the default status:
[image]
By the way, may i have your email? :)

zariiii9003, 2020-04-12:
There might be some way to do the channel assignments through the XL API but i have not looked into that. The usual workflow is to just do it via the vector hardware config.

I implemented a button for it in my application, which activates this method:
https://github.com/hardbyte/python-can/blob/2d672b340fc38f531a4a6470c8c2e288e5e50041/can/interfaces/vector/canlib.py#L581

CSG2019, 2020-04-14:
Hi zaiiii9003,
thank you for for the help.
actually, i am new to Python and I didn't find this function (def popup_vector_hw_configuration), so i don't know how to call it and have a check. :(, if you want, could you please show a simple example...

by the way, i search and find this method(can.detect_available_configs()), it seems that we can use this to find which config can be used.
but, after i tried to print, i got the below information. i am still a liitle confused, does it mean it has detected the Vector Tool?
[{'interface': 'virtual', 'channel': 'channel-3887'}, {'interface': 'vector', 'app_name': None, 'channel': 0, 'supports_fd': False}, {'interface': 'vector', 'app_name': None, 'channel': 1, 'supports_fd': False}, {'interface': 'vector', 'app_name': None, 'channel': 2, 'supports_fd': False}, {'interface': 'vector', 'app_name': None, 'channel': 5, 'supports_fd': False}, {'interface': 'vector', 'app_name': None, 'channel': 6, 'supports_fd': False}]

I think i have got the answer I wanted at the beginning.
thanks again. :)

zariiii9003, 2020-04-15:
`VectorBus.popup_vector_hw_configuration()` is only available in the development branch.

I do not know what that `detect_available_configs` does, never needed it.

CSG2019, 2020-04-17:
ok. thank you zariiii9003.

do you have any idea about ES581 interface?
i tried to run below code but there was an error:
ics.RuntimeError: Error: find_devices(): Failed to open library: 'icsneo40.dll' with error code: #126 
```
import can
from can.interfaces.ics_neovi.neovi_bus import NeoViBus

bus = NeoViBus(bustype='neovi', channel=1, bitrate=500000)
```

karlding, 2020-04-23:
In order to use the NeoVI interface, as part of `python_ics` installation, you also need to place the DLL in the system path. `python-can` does not handle installation of `python_ics`, per the `python-can` [documentation](https://python-can.readthedocs.io/en/3.3.2/interfaces/neovi.html#installation).

You should be able to follow the [installation instructions](https://python-ics.readthedocs.io/en/v4.2/#intrepid-icsneo40-library) for `python_ics` with respect to which DLLs need to be added to your system path.

karlding, 2020-09-19:
Closing this out as there doesn't seem to be anything actionable here. Please reopen if necessary.

closed 2020-09-19 by karlding