hardbyte/python-can #799: Record CAN BUS On GUI

CSG2019, 2020-03-21:
hi,

i want to create a small GUI to record the CAN bus log, but it failed in below code, i am new to the python, so,  if you know how to correct it , please help me . thank you very much.
i want to click 'start' to record the CAN bus log, and click the 'stop' to complete the record.

import can
from tkinter import *

def record_can_bus():
    can.Notifier(bus_channel_1, [can.Logger("test.asc"), can.Printer()])

def record_stop():
    can.Notifier.stop()

def app():
    window = Tk()
    window.title("Test")
    window.geometry('200x100+320+100')
    button_1 = Button(window, text='Start', command=record_can_bus)
    button_1.pack(pady=5)
    button_2 = Button(window, text='Stop', command=record_stop)
    button_2.pack(pady=5)
    button_3 = Button(window, text='Exit', command=window.destroy)
    button_3.pack(pady=5)
    window.mainloop()

if __name__ == "__main__":
    bus_channel_1 = can.interface.Bus(bustype='vector', app_name='CANoe', channel=[0])
    app()

CSG2019, 2020-03-27:
is there anybody can give me any suggestions? when i call the can.Notifier.stop(), the CAN BUS log didn't stop and i can find the asc file will continue to record...
thank you very much!

bessman, 2020-03-31:
> `can.Notifier(bus_channel_1, [can.Logger("test_2020.asc"), can.Printer()])`

This line creates a Notifier object, but you don't store it anywhere. That means you cannot stop it. You want to do something like `notifier = can.Notifier(bus_channel_1, [can.Logger("test_2020.asc"), can.Printer()])`. Then you can call `notifier.stop()` when you want to stop it.

> `can.Notifier.stop()`

This line calls `stop()` on the Notifier _class_, not the notifier object you previously created. Since `stop()` is not a class method, your script should stop here with TypeError.

CSG2019, 2020-04-03:
> This line creates a Notifier object, but you don't store it anywhere. That means you cannot stop it. You want to do something like `notifier = can.Notifier(bus_channel_1, [...]

hello, thank you for the response.
i tried to do like this `notifier = can.Notifier(bus_channel_1, [can.Logger("test_2020.asc"),` but when i called the stop function, it still can't stop it like i want to get, when i quit GUI， the can log file can be recorded.
so do have any better idea or suggestion to me?
thank you very much.

closed 2020-04-12 by CSG2019