hardbyte/python-can #800: No module named 'winreg'

skywo1f, 2020-03-31:
Traceback (most recent call last):
  File "receive_all.py", line 34, in <module>
    receive_all()
  File "receive_all.py", line 15, in receive_all
    bustype="pcan", channel="PCAN_USBBUS1", bitrate=250000
  File "/usr/local/lib/python3.7/dist-packages/python_can-3.2.0-py3.7.egg/can/interface.py", line 98, in __new__
  File "/usr/local/lib/python3.7/dist-packages/python_can-3.2.0-py3.7.egg/can/interface.py", line 40, in _get_class_for_interface
ImportError: Cannot import module can.interfaces.pcan for CAN interface 'pcan': No module named 'winreg'

bmeisels, 2020-03-31:
What operating system are you using?

skywo1f, 2020-04-01:
raspberry pi (raspbian)

christiansandberg, 2020-04-01:
Seems to be a real bug. Pcan should be supported on Linux as well but we assume that winreg is always present.

In general it is not so good to hack the basic.py file because it comes from Peak but I know we’ve made lots of changes so we will probably never be able to merge upstream changes anyway.

tamenol, 2020-04-02:
This could be my mistake from PR #724, commit https://github.com/hardbyte/python-can/pull/724/commits/6ed035cd5ef455bf2adf37b2c32c0cb435127afe

I added it on top without thinking about the Linux platform. Will make a fix

bmeisels, 2020-04-02:
According to the documentation https://www.peak-system.com/fileadmin/media/linux/index.htm ,
It seems that one should use socketcan for PCAN on linux. If @tamenol 's fix doesn't help maybe we should throw an exception on Linux suggesting to try socketcan.

tamenol, 2020-04-02:
I don't have a linux machine right now, but I would think the module `winreg` is not present on a Linux machine. The docs says its a MS Windows Specific Services ("modules that are only available on MS Windows platforms")
[https://docs.python.org/3/library/winreg.html](url)

bmeisels, 2020-04-02:
@tamenol I am sure you are right regarding winreg not being available. Regardless my comment was that PCAN probably wont work out of the box on Linux with PCAN basic and will work with socketcan

tamenol, 2020-04-02:
@bmeisels Oh, sorry, I misread. You probably right though, it can aslo explain why this bug only now surfaces.

PR #802 (2020-04-19): fix winreg bug in basic.py #800
only import winreg when on Windows

fixes #800
closed 2020-04-19 by hardbyte