hardbyte/python-can #803: BLF Parse error

yolaalabi, 2020-04-02:
hello, I get a BLF parse error when opening a normal BLF file, is there some workaround? can I email the file to somebody to check out?

hdnauth, 2020-04-07:
Hi, I have similar problem on parsing and logging as BLF. also, some option interface is needed for selecting channel among recorded data when more than one channel data is logged into one BLF file.

EbATpy, 2020-04-07:
#CAN FD   8< DLC has error

import can

file = r"C:\EbAT\blf files\Testing_w_FD"
log = can.BLFReader(file+'.blf')
for msg in log:
    print(msg)

Timestamp: 1585322863.337813        ID: 02eb    S                DLC:  5    00 00 00 01 00              Channel: 0
Timestamp: 1585322863.337939        ID: 02b2    S                DLC:  2    00 00                       Channel: 1
Timestamp: 1585322863.338042        ID: 02f5    S                DLC:  7    f5 00 00 00 00 00 07        Channel: 0
Timestamp: 1585322863.338055        ID: 02b8    S                DLC:  1    00                          Channel: 1
Timestamp: 1585322863.338229        ID: 0041    S     F BS EI    DLC:  8    ee 21 00 2c 10 6c 00 04     Channel: 2
Timestamp: 1585322863.338265        ID: 0305    S                DLC:  7    00 00 00 07 34 00 20        Channel: 0
Timestamp: 1585322863.338303        ID: 02d2    S                DLC:  8    95 00 00 00 00 00 00 00     Channel: 1
Timestamp: 1585322863.338360        ID: 007b    S     F BS       DLC:  8    80 0c 21 00 80 00 00 ff     Channel: 2
Timestamp: 1585322863.338417        ID: 0315    S                DLC:  3    33 00 00                    Channel: 0
Timestamp: 1585322863.338473        ID: 007d    S     F BS       DLC:  8    5f ff 80 01 80 02 40 e6     Channel: 2
Timestamp: 1585322863.338526        ID: 02d8    S                DLC:  7    00 00 03 3e 80 3e 80        Channel: 1
Timestamp: 1585322863.338603        ID: 0329    S                DLC:  5    94 00 00 00 00              Channel: 0
Timestamp: 1585322863.338668        ID: 007e    S     F BS       DLC: 24    00 00 00 00 07 00 00 00 00 00 00 00 00 00 00 00 00 70 00 00 00 00 00 00    Channel: 2
Timestamp: 1585322863.338758        ID: 02f8    S                DLC:  7    00 3e 80 00 3e 80 00        Channel: 1
Timestamp: 1585322863.338779        ID: 007f    S     F BS       DLC:  7    20 00 00 00 00 00 00        Channel: 2
Traceback (most recent call last):
  File "c:/Users/bz63yj/Desktop/Script1.py", line 10, in <module>
    for msg in log:
  File "C:\Python37_64\lib\site-packages\python_can-3.3.2-py3.7.egg\can\io\blf.py", line 195, in __iter__
can.io.blf.BLFParseError

christiansandberg, 2020-04-10:
There is a similar issue #786 for which there was a pull request made but was probably abandoned. I created a new which I tested with your attached log and had no issues. You can try it out yourself:

```
pip install https://github.com/hardbyte/python-can/archive/bugfix/803-blf-parse-error.zip
```

yolaalabi, 2020-04-15:
Hello Christian, that fixed my issue as well! Thank you, I appreciate this library and your support a lot.

PR #806 (2020-04-19): Fix BLFParseError
Seek for start of object instead of calculating it

Fixes #803
Fixes #786
commit 77802421 (2020-04-19): Seek for start of object instead of calculating it (#806)

Fixes #803
Fixes #786
closed 2020-04-19 by hardbyte