hardbyte/python-can #805: TestLoggerScript and TestPlayerScript test suites fail

Apteryks, 2020-04-02:
They are trying to launch "can_logger.py" or "can_player.py --help" (IS_UNIX == True), expecting these are in my PATH, but the program is not yet installed, so that fails.

Are these tests expected to be run after the software has been installed? That'd be weird, making those the only 2 out of 152 to have this requirement.

The error messages look like:
```
                else:
                    err_filename = orig_executable
                if errno_num != 0:
                    err_msg = os.strerror(errno_num)
                    if errno_num == errno.ENOENT:
                        err_msg += ': ' + repr(err_filename)
>               raise child_exception_type(errno_num, err_msg, err_filename)
E               FileNotFoundError: [Errno 2] No such file or directory: 'can_player.py': 'can_player.py'

/gnu/store/78w7y0lxar70j512iqw8x3nimzj10yga-python-3.7.4/lib/python3.7/subprocess.py:1522: FileNotFoundError
```

felixdivo, 2021-04-23:
This is still the case (on my local machines and the current `develop` branch).

felixdivo, 2022-02-06:
I can't reproduce this any more.

I did this in Docker (`docker run -it python:3.10 bash`) for the current `develop` branch:

```bash
cd home/
git clone https://github.com/hardbyte/python-can.git
cd python-can/
python setup.py install
pip install tox
tox -e gh  # github actions test suite
```

Only IPv6 stuff of the interface `udp_multicast` fails.

Also, skipping `python setup.py install` works.

In any case:

```
test/test_scripts.py::TestLoggerScript::test_do_commands_exist PASSED                          [ 80%]
test/test_scripts.py::TestLoggerScript::test_does_not_crash PASSED                             [ 81%]
test/test_scripts.py::TestPlayerScript::test_do_commands_exist PASSED                          [ 81%]
test/test_scripts.py::TestPlayerScript::test_does_not_crash PASSED                             [ 81%]
test/test_scripts.py::TestLogconvertScript::test_do_commands_exist PASSED                      [ 81%]
test/test_scripts.py::TestLogconvertScript::test_does_not_crash PASSED                         [ 82%]
```

Thus, this seems to be resolved.

closed 2022-02-06 by felixdivo