hardbyte/python-can #807: Build pyinstaller successful, but can‘t run the exe file

CSG2019, 2020-04-12:
Hi,
I used the pyinstaller to build the exe file, the exe file and related files can be created successfully.
But when i run the exe file, it showed that '' Failed to execute script...''.
in my .py file, i only import can and tkinter and time as below. and it can run normally in IDE.
`import can
from tkinter import *
import time
from tkinter import messagebox
`
Have you met the same porblem or give me any suggestions?

Thank you!

CSG2019, 2020-04-12:
by the way, in my code, my interface is Vertor.
in the warm txt file, it shows that there are some missing as below:

> 
missing module named fcntl - imported by can.interfaces.socketcan.socketcan (optional)
missing module named asyncio.DefaultEventLoopPolicy - imported by asyncio (delayed, conditional), asyncio.events (delayed, conditional)
missing module named _wrappers - imported by wrapt.wrappers (conditional, optional)
missing module named ConfigParser - imported by can.util (optional)
missing module named Queue - imported by can.listener (optional)
> 

thank you very much.

zariiii9003, 2020-04-12:
Run the .exe in console. It should print an error trace if you did not disable console in pyinstaller

CSG2019, 2020-04-12:
> Run the .exe in console. It should print an error trace if you did not disable console in pyinstaller

thank you for the suggestion.
after i run the .exe in console, the errow showed that No module named 'can.interface.vector'.
how to import this module? 
thank you.

zariiii9003, 2020-04-12:
either lookup up the hiddenimports setting of pyinstaller or try to import VectorBus directly:
`from can.interfaces.vector import VectorBus`

CSG2019, 2020-04-12:
> either lookup up the hiddenimports setting of pyinstaller or try to import VectorBus directly:
> `from can.interfaces.vector import VectorBus`

Hi, zariiii90003,
Thank you very much indeed for your quick reply.
As your suggestion, it can work now.

closed 2020-04-12 by CSG2019